dlmcpaul/EnphaseCollector #1: Comments

zjevik, 2019-01-27:
This is the best plug-and-play program to gather data from Envoy. I'm using the docker container and it works like a champ! (Make sure to add a link to it in your readme file)
Do you think you could share your Grafana dashboard? I saw it on whitpool.net.au and it looks wonderfully.

Thank you so much!

dlmcpaul, 2019-01-30:
Sorry, I am making changes to allow for multiple backends to recieve stats.  Add the below parameter to your docker run to get influxdb as a backend.  I will update the docs as well
 
-e SPRING_PROFILES_ACTIVE=influxdb

zjevik, 2019-01-30:
Oh, I see the PVOutput code, it would be nice if I could send the data to PVOutput more often than 15 min(using the Enlighten API). I'm looking forward to it.
Setting up the profile helped.