dlmcpaul/EnphaseCollector #3: History tab doesn't work with local database only

dgold9670, 2019-05-11:
I set up using local database only, and the GUI looks to work just fine on live view, but when I click 'history' it doesn't switch to that tab.

dlmcpaul, 2019-05-13:
The history tab is very much a work in progress at the moment.

I have made a build with the latest code and pushed to Docker so you can see what I am trying out.

At the moment the database is not exported from Docker so is not saved on upgrade though.

davegold24, 2019-05-13:
Pulled the current image (docker pull ...) and am seeing a few errors:
`2019-05-13 17:54:40.051 ERROR 1 --- [   scheduling-1] com.hz.services.OutputManager            : Failed to collect data from Enphase Controller - null

java.lang.NullPointerException: null
        at com.hz.models.envoy.json.Production.lambda$findBymeasurementType$1(Production.java:71) ~[classes!/:DEV]
        at java.base/java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:176) ~[na:na]
        at java.base/java.util.ArrayList$ArrayListSpliterator.tryAdvance(ArrayList.java:1631) ~[na:na]
        at java.base/java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:127) ~[na:na]
        at java.base/java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:502) ~[na:na]
        at java.base/java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:488) ~[na:na]
        at java.base/java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:474) ~[na:na]
        at java.base/java.util.stream.FindOps$FindOp.evaluateSequential(FindOps.java:150) ~[na:na]
        at java.base/java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234) ~[na:na]
        at java.base/java.util.stream.ReferencePipeline.findFirst(ReferencePipeline.java:543) ~[na:na]
        at com.hz.models.envoy.json.Production.findBymeasurementType(Production.java:72) ~[classes!/:DEV]
        at com.hz.models.envoy.json.Production.getProductionEim(Production.java:61) ~[classes!/:DEV]
        at com.hz.services.EnphaseService.systemNotReady(EnphaseService.java:146) ~[classes!/:DEV]
        at com.hz.services.EnphaseService.collectEnphaseData(EnphaseService.java:109) ~[classes!/:DEV]
        at com.hz.services.OutputManager.gather(OutputManager.java:46) ~[classes!/:DEV]
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[na:na]
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[na:na]
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:na]
        at java.base/java.lang.reflect.Method.invoke(Method.java:566) ~[na:na]
        at org.springframework.scheduling.support.ScheduledMethodRunnable.run(ScheduledMethodRunnable.java:84) ~[spring-context-5.1.6.RELEASE.jar!/:5.1.6.RELEASE]
        at org.springframework.scheduling.support.DelegatingErrorHandlingRunnable.run(DelegatingErrorHandlingRunnable.java:54) ~[spring-context-5.1.6.RELEASE.jar!/:5.1.6.RELEASE]
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515) ~[na:na]
        at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) ~[na:na]
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) ~[na:na]
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[na:na]
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[na:na]
        at java.base/java.lang.Thread.run(Thread.java:834) ~[na:na]`

I'm running it via:
`[root@dg-test ~]# cat enphasecollector.sh
docker run \
-e TZ=EST \
-e ENVOY_CONTROLLER_PASSWORD=012345 \
-e ENVOY_CONTROLLER_HOST=192.168.2.158 \
-p 8080:8080 \
dlmcpaul/enphasecollector`
(where 012345 is replaced by my actual password)

I also manually deleted the container and image, just in case docker pull was an issue...same thing happened.

I did take a look at the tabs, and I like your idea.
I did notice on the previous build that it was reporting consumption, even though my model only reports on consumption. If you want any outputs, let me know.

dlmcpaul, 2019-05-13:
Oh ok, no production information is available.

Hmm could you send me the output of 

http://192.168.2.158/home.json
http://192.168.2.158/production.json?details=1

and I will see if I can handle that better.

davegold24, 2019-05-13:
It is night and rainy, so...
home.json:
`{"software_build_epoch":1464913042,"is_nonvoy":false,"db_size":"14` `MB","db_percent_full":"3","timezone":"US/Eastern","current_date":"05/13/2019","current_time":"19:55","network":{"web_comm":true,"ever_reported_to_enlighten":true,"last_enlighten_report_time":1557774153,"primary_interface":"ppp0","interfaces":[{"signal_strength":25,"signal_strength_max":31,"network":true,"type":"cellular","interface":"ppp0","dhcp":true,"ip":"10.44.102.163","carrier":true},{"type":"ethernet","interface":"eth0","mac":"00:1D:C0:66:6E:C5","dhcp":true,"ip":"169.254.120.1","signal_strength":0,"signal_strength_max":1,"carrier":false},{"signal_strength":3,"signal_strength_max":5,"type":"wifi","interface":"wlan0","mac":"68:9E:19:19:3E:84","dhcp":true,"ip":"192.168.2.158","carrier":true,"supported":true,"present":true,"configured":true,"status":"connected"}]},"comm":{"num":20,"level":5},"alerts":[],"update_status":"satisfied"}`

production.json?details=1:
`{"production":[{"type":"inverters","wNow":1,"whLifetime":12653472.193611111,"readingTime":1557791836,"activeCount":20},{"type":"eim","activeCount":1,"whLifetime":12605195.311,"whLastSevenDays":158680.311,"whToday":13499.311,"wNow":-1.707,"rmsCurrent":2.322,"rmsVoltage":242.506,"reactPwr":272.909,"apprntPwr":281.518,"pwrFactor":-0.01,"readingTime":1557791836}],"consumption":[{"type":"eim","activeCount":1,"whLifetime":12604234.122,"whLastSevenDays":158671.122,"whToday":13498.122,"wNow":-1.707,"varhLeadToday":1.195,"varhLagToday":4704.803,"vahToday":683.805,"varhLeadLifetime":4742.195,"varhLagLifetime":6193093.803,"vahLifetime":530587.805,"rmsCurrent":2.599,"rmsVoltage":242.49,"reactPwr":272.49,"apprntPwr":630.211,"pwrFactor":-0.0,"readingTime":1557791836}]}`

dlmcpaul, 2019-05-14:
Ok, I have pushed another image that should handle the missing fields from your enphase controller.

Looks like yours is an older model/Version.

davegold24, 2019-05-14:
Mine was installed August 2016. Software version is D4.2.27 (b13066), software build date is Jun 02, 2016 8:17 PM.
New version comes up and works nicely. If there is anything you'd like me to test, let me know. And I appreciate your code updates!

dlmcpaul, 2019-05-15:
Glad to see that worked.  My Software is D4.5.79 so a few versions on and looks like they added a lot more properties.

Might also be useful to get these as well to see if there are any other differences
http://192.168.2.158/inventory.json?deleted=1
http://192.168.2.158/info.xml

The first should ask for user/password.  user will be envoy and the password you know.

dgold9670, 2019-05-15:
output of first:
`[{"type":"PCU","devices":[{"part_num":"800-00350-r02","installed":"1471983371","serial_num":"121603070495","device_status":["envoy.global.ok"],"last_rpt_date":"1557924245","admin_state":2,"created_date":"1471983371","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627390225,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983375","serial_num":"121603072813","device_status":["envoy.global.ok"],"last_rpt_date":"1557924247","admin_state":2,"created_date":"1471983375","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627390481,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983378","serial_num":"121603073673","device_status":["envoy.global.ok"],"last_rpt_date":"1557924223","admin_state":2,"created_date":"1471983378","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627390993,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983388","serial_num":"121603072754","device_status":["envoy.global.ok"],"last_rpt_date":"1557924225","admin_state":2,"created_date":"1471983388","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627391249,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983392","serial_num":"121603067704","device_status":["envoy.global.ok"],"last_rpt_date":"1557924227","admin_state":2,"created_date":"1471983392","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627391505,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983396","serial_num":"121603072758","device_status":["envoy.global.ok"],"last_rpt_date":"1557924227","admin_state":2,"created_date":"1471983396","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627391761,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983415","serial_num":"121603073085","device_status":["envoy.global.ok"],"last_rpt_date":"1557924229","admin_state":2,"created_date":"1471983415","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627392017,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983418","serial_num":"121603067453","device_status":["envoy.global.ok"],"last_rpt_date":"1557924230","admin_state":2,"created_date":"1471983418","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627392273,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983420","serial_num":"121603067248","device_status":["envoy.global.ok"],"last_rpt_date":"1557924230","admin_state":2,"created_date":"1471983420","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627392529,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983423","serial_num":"121603064274","device_status":["envoy.global.ok"],"last_rpt_date":"1557924232","admin_state":2,"created_date":"1471983423","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627392785,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983454","serial_num":"121603073538","device_status":["envoy.global.ok"],"last_rpt_date":"1557924233","admin_state":2,"created_date":"1471983454","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627393041,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983463","serial_num":"121603050803","device_status":["envoy.global.ok"],"last_rpt_date":"1557924234","admin_state":2,"created_date":"1471983463","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627393297,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983466","serial_num":"121603073339","device_status":["envoy.global.ok"],"last_rpt_date":"1557924236","admin_state":2,"created_date":"1471983466","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060-r01-v01.22.00","producing":true,"communicating":true,"chaneid":1627393553,"device_control":[{"gficlearset":false}]},{"part_num":"800-00350-r02","installed":"1471983468","serial_num":"121603073552","device_status":["envoy.global.ok"],"last_rpt_date":"1557924237","admin_state":2,"created_date":"1471983468","img_load_date":"1424997875","img_pnum_running":"520-00044-r01-v01.22.00","ptpn":"540-00060- [... cut ...]
    02.00.00
    922
  
  
    500-00001-r01
    02.00.00
    922
  
  
    500-00011-r01
    04.00.00
    26b550
  
  
    590-00018-r01
    02.00.01
    1d81f5
  
  
    500-00002-r01
    04.02.27
    b13066
  
  
    500-00004-r01
    01.01.41
    00ecd8
  
  
    500-00008-r01
    01.05.07
    eb932f
  
  
    500-00010-r01
    04.02.43
    a3ade0
  
  
    500-00013-r01
    02.01.04
    299acc
  
  
    500-00012-r01
    01.00.00
    b13066
  
  
    500-00016-r01
    02.00.00
    54a6dc
  
`

dgold9670, 2019-05-18:
The weekly tab is really showing daily info. Could I suggest renaming it to "Live Daily Weekly Monthly"?
Also, Daily, Weekly and Monthly all show the same. Although I suspect you already knew that.

To other items.
First, could I suggest that you note the update time on the right pane?
Second, could you display when there is an error pulling data? Here is the error the logs show me:
2019-05-18 14:55:54.111 ERROR 1 --- [   scheduling-1] com.hz.services.EnphaseService           : Failed to retrieve Solar stats. Exception was I/O error on GET request for "http://192.168.2.158/home.json": Bad chunk header: <!DOCTYPE html PUBLIC "-//IETF//DTD HTML 2.0//EN">; nested exception is org.apache.http.MalformedChunkCodingException: Bad chunk header: <!DOCTYPE html PUBLIC "-//IETF//DTD HTML 2.0//EN">

dlmcpaul, 2019-05-19:
Yeah, still trying to work out what and how I want to display historical data

I have added the the time the controller is polled and it should also mention if there was an error.  Will push a new version sometime.  Still testing some other changes.

dlmcpaul, 2019-08-08:
History tabs now update.

closed 2019-08-08 by dlmcpaul