dlmcpaul/EnphaseCollector #4: How to compile/run outside docker?

lincolnturner, 2019-09-06:
This looks like a great project and I'd like to use it with my enphase system.
But no docker on my device.

Any hints on how to run on a native JRE? mvn install doesn't work.

dlmcpaul, 2019-09-10:
"mvn package" should produce a jar file in the target directory called enphasecollector-DEV.jar

then "java -jar enphasecollector-DEV.jar" will start it.  Java 11 is needed to run.

You will still need to set the environment parameters to configure it to your system.

You will need at least these 2 set.
ENVOY_CONTROLLER_HOST
ENVOY_CONTROLLER_PASSWORD

Running a jar as a service depends on your operating system.

lincolnturner, 2019-09-10:
Thanks very much.

I've "discovered" the http://envoy/stream/meter interface which gives 1s
updates on overall production and consumption, so am hacking that into a
pgsql timescaledb at present. But I may well come back to yoru java...

Best,

Lincoln

-- 
*Lincoln Turner*
Senior Lecturer
Fellow - Monash Education Academy

I am an LGBTIQ Ally <http://monash.edu/ally>

*School of Physics & Astronomy*
Monash University
G.06, New Horizons Centre
20 Research Way, Clayton Campus
Victoria 3800, Australia

T: +61 3 9905 1726
*bec.physics.monash.edu.au <http://bec.physics.monash.edu.au>*

dlmcpaul, 2019-10-29:
Modified readme to add running directly with java

closed 2019-10-29 by dlmcpaul