dlmcpaul/EnphaseCollector #5: Data is showing on the graph, but not in the text fields on the right

ErikLentz, 2020-06-11:
You can see my setup here: https://dev1.eriklentz.com/solar/

I'm running the Java version of the collector on Debian Linux.

I stopped the collector, added this flag to my run command to change from the default polling interval: --envoy.refreshseconds=15000 
Then started the collector again. As far as what I intended to accomplish, it works, it now polls more often.

However, the graph still displays the collected data, but most of the text fields on the right are empty and all historical data cannot be displayed.

Reverting to the default polling interval does not appear to correct the issue. Has the internal database become corrupted?

ErikLentz, 2020-06-11:
Nevermind, removing the polling interval flag entirely fixed the problem. Is it not possible to change the polling interval, or will it require a clean internal database to do so?

dlmcpaul, 2020-06-14:
I will see what I can discover about that but changing the refresh seconds should not affect the text field display.

It could affect the daily rollup of data to the history table as it needs to convert all the daily values to kWh.

dlmcpaul, 2020-06-15:
The calculations for costs and displays look to work on RefreshSeconds being a multiple of 60000

I have added a change to allow for fractions of 60000 and will test further.

The history calculations assume that the RefreshSeconds value does not change over the period so some cleanup in the database might be needed once the fix is released.  Should be simple enough just need to multiply some values in the summary table I think.