dlmcpaul/EnphaseCollector #6: How to run on Android Device

smacinto, 2020-07-06:
Thanks David for such a great app. 

I would like to run EnphaseCollector on an android device but am unsure if this is possible? Any tips would be much appreciated.

dlmcpaul, 2020-07-07:
There is not likely to be much I can do to run this app on android.  Obviously you can display the web page on an android device but you will need a machine with java or docker to run the application.

Android does not seem to have either.  I did manage to install QEMU and try running java in a alpine image but not a lot of luck.

I would like to know more about your situation.  Is it because you don't have a machine that is on all the time?

smacinto, 2020-07-07:
David I was hoping to run your application on an old android galaxy tablet and have it displayed in the kitchen so that we could monitor usage and production every minute. I find MyEnlighten not granular enough for identifying devices consuming power. I do have a desktop and laptop but don't wish to run these devices 24*7.

dlmcpaul, 2020-07-07:
Yes I understand.  I have a Synology NAS drive that needs to be on 24*7 and it runs docker images so I host there.

dlmcpaul, 2020-10-17:
I recently added a ARM docker image for use with raspberry pi devices

smacinto, 2021-07-17:
Thanks David. This works a treat on my raspberry and I am able to view the graphs on my old galaxy tablet as hoped.

smacinto, 2022-07-05:
Hi David. Do you have an updated ARM docker image for a raspberry pi? I have tried running the latest 0.25 version with the docker platform option without success.

dlmcpaul, 2022-07-05:
I have pushed the 0.25 release to the dlmcpaul/enphasecollector-arm image

Let me know how it goes as I cannot test arm images.

smacinto, 2022-07-05:
Hi David,

It would appear the enphasecollector-arm image successfully downloaded but I tried running with and without the docker
platform option without success.

Any ideas?

Thanks for your great app and prompt response. 

Regards Stuart.

	pi@raspberrypi:~ $ docker images
	REPOSITORY                  TAG       IMAGE ID       CREATED         SIZE
	dlmcpaul/enphasecollector   latest    6c2428160f71   9 days ago      281MB
	dlmcpaul/enphasecollector   arm       deaaca08f0b0   22 months ago   467MB
	hello-world                 latest    851163c78e4a   2 years ago     4.85kB
	pi@raspberrypi:~ $ 
	pi@raspberrypi:~ $ docker pull dlmcpaul/enphasecollector-arm
	Using default tag: latest
	latest: Pulling from dlmcpaul/enphasecollector-arm
	11e23ac719b3: Pull complete 
	68e4a04805fb: Pull complete 
	e41ee0d8f3dc: Pull complete 
	bdd3adec87ea: Pull complete 
	93f308e998d2: Pull complete 
	3911327e6f99: Pull complete 
	2d21732c3c14: Pull complete 
	Digest: sha256:18e96f93baac2608d63caca0c9f50ee5648e97ce0feb357422fa9a165f1568e8
	Status: Downloaded newer image for dlmcpaul/enphasecollector-arm:latest
	docker.io/dlmcpaul/enphasecollector-arm:latest
	pi@raspberrypi:~ $ 
	pi@raspberrypi:~ $ docker images
	REPOSITORY                      TAG       IMAGE ID       CREATED         SIZE
	dlmcpaul/enphasecollector-arm   latest    4bab9bc9a27f   9 days ago      333MB
	dlmcpaul/enphasecollector       latest    6c2428160f71   9 days ago      281MB
	dlmcpaul/enphasecollector       arm       deaaca08f0b0   22 months ago   467MB
	hello-world                     latest    851163c78e4a   2 years ago     4.85kB
	pi@raspberrypi:~ $ 

	

	pi@raspberrypi:~ $ docker run \
	> --mount source=enphase,target=/internal_db \
	> -e TZ=Australia/Sydney \
	> -e ENVOY_CONTROLLER_PASSWORD=*** \
	> -e ENVOY_CONTROLLER_HOST=*** \
	> -e ENVOY_REFRESHSECONDS=15000 \
	> -e ENVOY_PAYMENTPERKILOWATT=0.102 \
	> -e ENVOY_CHARGEPERKILOWATT=0.288 \
	> -e ENVOY_DAILYSUPPLYCHARGE=1.31 \
	> -e ENVOY_BANDS[0].FROM=0700 \
	> -e ENVOY_BANDS[0].TO=0859 \
	> -e ENVOY_BANDS[0].COLOUR=#FA8072 \
	> -e ENVOY_BANDS[1].FROM=1700 \
	> -e ENVOY_BANDS[1].TO=1959 \
	> -e ENVOY_BANDS[1].COLOUR=#FA8072 \
	> -e ENVOY_BANDS[2].FROM=0900 \
	> -e ENVOY_BANDS[2].TO=1659 \
	> -e ENVOY_BANDS[2].COLOUR=#F0E68C \
	> -e ENVOY_BANDS[3].FROM=2000 \
	> -e ENVOY_BANDS[3].TO=2159 \
	> -e ENVOY_BANDS[3].COLOUR=#F0E68C \
	> -e ENVOY_BANDS[4].FROM=2200 \
	> -e ENVOY_BANDS[4].TO=0659 \
	> -e ENVOY_BANDS[4].COLOUR=#87CEEB \
	> -p 8080:8080 \
	> dlmcpaul/enphasecollector-arm:latest
	WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
	standard_init_linux.go:219: exec user process caused: exec format error
	
[... 155 lines cut ...]
	> -p 8080:8080 \
	> dlmcpaul/enphasecollector-arm:latest 
	WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
	9e415c417c1f4af6db3fa2251375e4a8df9ffd8617a873a655a919c593ee7f11
	pi@raspberrypi:~ $ 

	pi@raspberrypi:~ $ docker container ls
	CONTAINER ID   IMAGE                                  COMMAND                  CREATED         STATUS                         PORTS     NAMES
	9e415c417c1f   dlmcpaul/enphasecollector-arm:latest   "java -Djava.securit\u2026"   2 minutes ago   Restarting (1) 4 seconds ago             serene_darwin
	
	pi@raspberrypi:~ $ !-1
	docker container ls
	CONTAINER ID   IMAGE                                  COMMAND                  CREATED         STATUS                          PORTS     NAMES
	9e415c417c1f   dlmcpaul/enphasecollector-arm:latest   "java -Djava.securit\u2026"   2 minutes ago   Restarting (1) 18 seconds ago             serene_darwin
	
	pi@raspberrypi:~ $ !!
	docker container ls
	CONTAINER ID   IMAGE                                  COMMAND                  CREATED         STATUS                         PORTS     NAMES
	9e415c417c1f   dlmcpaul/enphasecollector-arm:latest   "java -Djava.securit\u2026"   3 minutes ago   Restarting (1) 9 seconds ago             serene_darwin
	
	pi@raspberrypi:~ $ !!
	docker container ls
	CONTAINER ID   IMAGE                                  COMMAND                  CREATED         STATUS                          PORTS     NAMES
	9e415c417c1f   dlmcpaul/enphasecollector-arm:latest   "java -Djava.securit\u2026"   4 minutes ago   Restarting (1) 58 seconds ago             serene_darwin
	pi@raspberrypi:~ $

smacinto, 2022-07-06:
Hi David,

Last night I forced the upload of the new image and restarted.  enphasecollector-arm was running but not producing any output when I browse to the webpage??
I'm not sure how to debug docker - any ideas?

Thanks Stuart.

	pi@raspberrypi:~ $ docker pull dlmcpaul/enphasecollector-arm
	Using default tag: latest
	latest: Pulling from dlmcpaul/enphasecollector-arm
	36754881d0ec: Already exists 
	c1a22ec594d8: Already exists 
	9bc765178326: Already exists 
	2965aa4bc865: Already exists 
	93f308e998d2: Already exists 
	62fd43bae890: Already exists 
	39e2200e07ab: Already exists 
	Digest: sha256:51f844c20456cfd7a9f9fa715dc853d3d583529eed2f2e3544ddafbb6a05f028
	Status: Downloaded newer image for dlmcpaul/enphasecollector-arm:latest
	docker.io/dlmcpaul/enphasecollector-arm:latest
	pi@raspberrypi:~ $ 
	pi@raspberrypi:~ $ 
	pi@raspberrypi:~ $ docker rm -vf $(docker ps -a -q)
	71f9b1999ae7
	0f06e37ba2db
	0f9889a80454
	78de88394120
	pi@raspberrypi:~ $ docker rmi -f $(docker images -a -q)
	Untagged: dlmcpaul/enphasecollector-arm:latest
	Untagged: dlmcpaul/enphasecollector-arm@sha256:51f844c20456cfd7a9f9fa715dc853d3d583529eed2f2e3544ddafbb6a05f028
	Deleted: sha256:66ce280632c0a71449b5e3cc75d43e503014424039e8e3c76d409fb8ac8e0bf0
	Deleted: sha256:971c9631811d834e5d20cc2eafaa5a4563a06e4d7311453df3c40006caf27dd0
	Deleted: sha256:f07520dab6de8462a641c8ef1e4043da54db9418fdc8b73b5eb0ae8a80dde6b9
	Deleted: sha256:0dd7dc98eff41e3c94d7ed4d75f69a417ee360d57c473775ca7ca54bb5deb14e
	Deleted: sha256:0bb647ee62dda0eefafc838a9d468f1c2740d2b44774f2d4ebe444d565f56c83
	Deleted: sha256:7a57c79d51dc2231350b540e03b7063975736b246ff2d3d239b3f6495eeedc14
	Deleted: sha256:b9e493c93eba538c37f585e653ae4d534284194a13e290277a8ed80ae601f1b6
	Deleted: sha256:9a56bec5e307ff2919e3a716e44effe4f6cf54a8fbac14e24d4d52bc27085f13
	Untagged: dlmcpaul/enphasecollector:arm
	Untagged: dlmcpaul/enphasecollector@sha256:6c926d38d80c91f9de0e675d70f8b85863d87809ca880c5bbb5b12faafc9ace5
	Deleted: sha256:deaaca08f0b07b2b23c9f5f1b80bdacf537888d789150adab2bbe329966e4527
	Deleted: sha256:518b535466dd300793c3a1d2824791a15be05dd73bd4ea892cd0fe3b5ffb9d55
	Deleted: sha256:78a14ba642243d56546d02284b5a762b9f3947f7f06a0b5605d198a232def807
	Deleted: sha256:560bfa762d4fe23308750e96582d3ad721a2a519e1077a6d60073c13b443c212
	Deleted: sha256:a6ab3ff46ae547142883356a8476d93ea4c6f101dab771cf3a1dbee48271aec8
	Deleted: sha256:e18b30a0e515d54707fbb2cd4310bca8dc15f947645c2634c833c4ac8014eedc
	Deleted: sha256:c624189ba4a0d8f909568511e309e238161a02499e6d9cf5ed42e87f4e671819
	Deleted: sha256:5bd86e6d94cb2880e4ca820c380db22984327fadf6acd5d93dae9afe561091ef
	Untagged: hello-world:latest
	Untagged: hello-world@sha256:f2266cbfc127c960fd30e76b7c792dc23b588c0db76233517e1891a4e357d519
	Deleted: sha256:851163c78e4ad68e6fe5391f0894aafd164d40c4d4d0a56b4291f0dc2c75cc2c
	Deleted: sha256:2536d8d4e4b1baa6515d44eb77a1402d6be0a533e7d191c51cb8428ba5ece3f4
	pi@raspberrypi:~ $ docker system prune
	WARNING! This will remove:
	  - all stopped containers
	  - all networks not used by at least one container
	  - all dangling images
	  - all dangling build cache

	Are you sure you want to continue? [y/N] y
	Total reclaimed space: 0B
	pi@raspberrypi:~ $ docker images
	REPOSITORY   TAG       IMAGE ID   CREATED   SIZE
	pi@raspberrypi:~ $ 
	pi@raspberrypi:~ $ docker pull dlmcpaul/enphasecollector-arm
	Using default tag: latest
	latest: Pulling from dlmcpaul/enphasecollector-arm
	36754881d0ec: Pull complete 
	c1a22ec594d8: Pull complete 
	9bc765178326: Pull complete 
	2965aa4bc865: Pull complete 
	93f308e998d2: Pull complete 
	62fd43bae890: Pull complete 
	39e2200e07ab: Pull complete 
	Digest: sha256:51f844c20456cfd7a9f9fa715dc853d3d583529eed2f2e3544ddafbb6a05f028
	Status: Downloaded newer image for dlmcpaul/enphasecollector-arm:latest
	docker.io/dlmcpaul/enphasecollector-arm:latest
	pi@raspberrypi:~ $ 

	pi@raspberrypi:~ $ docker container ls -a
	CONTAINER ID   IMAGE                                  COMMAND                  CREATED        STATUS        PORTS                                       NAMES
	18ce9f23f8f4   dlmcpaul/enphasecollector-arm:latest   "java -Djava.securit\u2026"   18 hours ago   Up 18 hours   0.0.0.0:8080->8080/tcp, :::8080->8080/tcp   charming_nightingale
	pi@raspberrypi:~ $ 

Latest syntax:

	pi@raspberrypi:~ $ docker run -d --restart unless-stopped --mount source=enphase,target=/internal_db \
	> -e TZ=Australia/Sydney \
	> -e ENVOY_CONTROLLER_PASSWORD=*** \
	> -e ENVOY_CONTROLLER_HOST=*** \
	> -e ENVOY_REFRESHSECONDS=15000 \
	> -e ENVOY_PAYMENTPERKILOWATT=0.102 \
	> -e ENVOY_CHARGEPERKILOWATT=0.288 \
	> -e ENVOY_DAILYSUPPLYCHARGE=1.31 \
	> -e ENVOY_BANDS[0].FROM=0700 \
	> -e ENVOY_BANDS[0].TO=0859 \
	> -e ENVOY_BANDS[0].COLOUR=#FA8072 \
	> -e ENVOY_BANDS[1].FROM=1700 \
	> -e ENVOY_BANDS[1].TO=1959 \
	> -e ENVOY_BANDS[1].COLOUR=#FA8072 \
	> -e ENVOY_BANDS[2].FROM=0900 \
	> -e ENVOY_BANDS[2].TO=1659 \
	> -e ENVOY_BANDS[2].COLOUR=#F0E68C \
	> -e ENVOY_BANDS[3].FROM=2000 \
	> -e ENVOY_BANDS[3].TO=2159 \
	> -e ENVOY_BANDS[3].COLOUR=#F0E68C \
	> -e ENVOY_BANDS[4].FROM=2200 \
	> -e ENVOY_BANDS[4].TO=0659 \
	> -e ENVOY_BANDS[4].COLOUR=#87CEEB \
	> -p 8080:8080 \
	> dlmcpaul/enphasecollector-arm:latest 
	4e35567e1577c7edcf0fa648ca64b318c42105f3a3109fad23f5d8b16e01092f
	pi@raspberrypi:~ $

dlmcpaul, 2022-07-06:
docker logs <containerid> 

This should output the logs from the app.  Might want to pipe to a file as it might be large if lots of errors

smacinto, 2022-07-07:
Hi David,

I tried running "docker logs" whilst running the enphasecollector-arm:latest container but didn't get any output. I reinstalled the original enphasecollector:arm container and it provides STDOUT output and also provides debug output from the "docker logs" command. I don't get any STDOUT output at all from enphasecollector-arm:latest container - No Spring Boot messages just nothing.

It's as if the program starts but doesn't do anything?? I have included the STDOUT output from enphasecollector:arm just in case it provides any clues.

Thanks Stuart.

docker run --mount source=enphase,target=/internal_db -e TZ=Australia/Sydney -e ENVOY_CONTROLLER_PASSWORD=*** -e ENVOY_CONTROLLER_HOST=*** -e ENVOY_REFRESHSECONDS=15000 -e ENVOY_PAYMENTPERKILOWATT=0.102 -e ENVOY_CHARGEPERKILOWATT=0.288 -e ENVOY_DAILYSUPPLYCHARGE=1.31 -e ENVOY_BANDS[0].FROM=0700 -e ENVOY_BANDS[0].TO=0859 -e ENVOY_BANDS[0].COLOUR=#FA8072 -e ENVOY_BANDS[1].FROM=1700 -e ENVOY_BANDS[1].TO=1959 -e ENVOY_BANDS[1].COLOUR=#FA8072 -e ENVOY_BANDS[2].FROM=0900 -e ENVOY_BANDS[2].TO=1659 -e ENVOY_BANDS[2].COLOUR=#F0E68C -e ENVOY_BANDS[3].FROM=2000 -e ENVOY_BANDS[3].TO=2159 -e ENVOY_BANDS[3].COLOUR=#F0E68C -e ENVOY_BANDS[4].FROM=2200 -e ENVOY_BANDS[4].TO=0659 -e ENVOY_BANDS[4].COLOUR=#87CEEB -p 8080:8080 dlmcpaul/enphasecollector:arm

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::        (v2.2.9.RELEASE)

2022-07-07 17:19:47.560  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : Starting EnphaseCollectorApplication vdevelopment-SNAPSHOT on e244ec8258a6 with PID 1 (/app.jar started by root in /)
2022-07-07 17:19:47.577  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : No active profile set, falling back to default profiles: default
2022-07-07 17:19:48.873  INFO 1 --- [           main] o.s.c.a.ConfigurationClassParser         : Properties location [classpath:/release.properties] not resolvable: class path resource [release.properties] cannot be opened because it does not exist
2022-07-07 17:19:53.632  INFO 1 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Bootstrapping Spring Data JPA repositories in DEFAULT mode.
2022-07-07 17:19:54.119  INFO 1 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Finished Spring Data repository scanning in 439ms. Found 4 JPA repository interfaces.
2022-07-07 17:19:55.832  INFO 1 --- [           main] trationDelegate$BeanPostProcessorChecker : Bean 'org.springframework.ws.config.annotation.DelegatingWsConfiguration' of type [org.springframework.ws.config.annotation.DelegatingWsConfiguration$$EnhancerBySpringCGLIB$$204e84ba] is not eligible for getting processed by all BeanPostProcessors (for example: not eligible for auto-proxying)
2022-07-07 17:19:56.094  INFO 1 --- [           main] .w.s.a.s.AnnotationActionEndpointMapping : Supporting [WS-Addressing August 2004, WS-Addressing 1.0]
2022-07-07 17:19:58.352  INFO 1 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat initialized with port(s): 8080 (http)
2022-07-07 17:19:58.421  INFO 1 --- [           main] o.apache.catalina.core.StandardService   : Starting service [Tomcat]
2022-07-07 17:19:58.423  INFO 1 --- [           main] org.apache.catalina.core.StandardEngine  : Starting Servlet engine: [Apache Tomcat/9.0.37]
2022-07-07 17:19:58.758  INFO 1 --- [           main] o.a.c.c.C.[Tomcat].[localhost].[/solar]  : Initializing Spring embedded WebApplicationContext
2022-07-07 17:19:58.759  INFO 1 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 10844 ms
2022-07-07 17:20:00.892  INFO 1 --- [           main] o.hibernate.jpa.internal.util.LogHelper  : HHH000204: Processing PersistenceUnitInfo [name: default]
2022-07-07 17:20:01.517  INFO 1 --- [           main] org.hibernate.Version                    : HHH000412: Hibernate ORM core version 5.4.18.Final
2022-07-07 17:20:01.527  INFO 1 --- [           main] org.hibernate.cfg.Environment            : HHH000205: Loaded properties from resource hibernate.properties: {hibernate.bytecode.use_reflection_optimizer=false, hibernate.bytecode.provider=none}
2022-07-07 17:20:01.723  INFO 1 --- [           main] o.hibernate.annotations.common.Version   : HCANN000001: Hibernate Commons Annotations {5.1.0.Final}
2022-07-07 17:20:02.383  INFO 1 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Starting...
2022-07-07 17:20:03.497  INFO 1 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Start completed.
2022-07-07 17:20:03.715  INFO 1 --- [           main] org.hibernate.dialect.Dialect            : HHH000400: Using dialect: org.hibernate.dialect.H2Dialect
2022-07-07 17:20:06.949  INFO 1 --- [           main] o.h.e.t.j.p.i.JtaPlatformInitiator       : HHH000490: Using JtaPlatform implementation: [org.hibernate.engine.transaction.jta.platform.internal.NoJtaPlatform]
2022-07-07 17:20:07.002  INFO 1 --- [           main] j.LocalContainerEntityManagerFactoryBean : Initialized JPA EntityManagerFactory for persistence unit 'default'
2022-07-07 17:20:07.335  INFO 1 --- [           main] c.h.c.EnphaseRestClientConfig            : Reading from insecure Envoy controller endpoint http://192.168.20.3
[... cut ...]
2022-07-07 17:21:17.096  INFO 1 --- [extShutdownHook] o.s.s.concurrent.ThreadPoolTaskExecutor  : Shutting down ExecutorService 'applicationTaskExecutor'
2022-07-07 17:21:17.106  INFO 1 --- [extShutdownHook] j.LocalContainerEntityManagerFactoryBean : Closing JPA EntityManagerFactory for persistence unit 'default'
2022-07-07 17:21:17.113  INFO 1 --- [extShutdownHook] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Shutdown initiated...
2022-07-07 17:21:17.123  INFO 1 --- [extShutdownHook] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Shutdown completed.
pi@raspberrypi:~ $

dlmcpaul, 2022-07-11:
I have pushed another arm version dlmcpaul/enphasecollector-arm:0.25

Try with the -it parameters to see what it logs

smacinto, 2022-07-11:
Hi David,

It successfully downloaded but doesn't want to run with or without the "-it" option.

Thanks for your ongoing assistance.

	pi@raspberrypi:~ $ docker run \
	> --mount source=enphase,target=/internal_db \
	> -e TZ=Australia/Sydney \
	> -e ENVOY_CONTROLLER_PASSWORD=*** \
	> -e ENVOY_CONTROLLER_HOST=**** \
	> -e ENVOY_REFRESHSECONDS=15000 \
	> -e ENVOY_PAYMENTPERKILOWATT=0.102 \
	> -e ENVOY_CHARGEPERKILOWATT=0.288 \
	> -e ENVOY_DAILYSUPPLYCHARGE=1.31 \
	> -e ENVOY_BANDS[0].FROM=0700 \
	> -e ENVOY_BANDS[0].TO=0859 \
	> -e ENVOY_BANDS[0].COLOUR=#FA8072 \
	> -e ENVOY_BANDS[1].FROM=1700 \
	> -e ENVOY_BANDS[1].TO=1959 \
	> -e ENVOY_BANDS[1].COLOUR=#FA8072 \
	> -e ENVOY_BANDS[2].FROM=0900 \
	> -e ENVOY_BANDS[2].TO=1659 \
	> -e ENVOY_BANDS[2].COLOUR=#F0E68C \
	> -e ENVOY_BANDS[3].FROM=2000 \
	> -e ENVOY_BANDS[3].TO=2159 \
	> -e ENVOY_BANDS[3].COLOUR=#F0E68C \
	> -e ENVOY_BANDS[4].FROM=2200 \
	> -e ENVOY_BANDS[4].TO=0659 \
	> -e ENVOY_BANDS[4].COLOUR=#87CEEB \
	> -p 8080:8080 \
	> -it dlmcpaul/enphasecollector-arm:0.25
	standard_init_linux.go:219: exec user process caused: exec format error
	pi@raspberrypi:~ $

smacinto, 2022-07-12:
Hi David,

Reran with the latest version and got the same result. The command "uname -m" reports armv7l. No problems me trying to build on the rasberrypi if you can give me a few pointers.

Thanks Stuart.

pi@raspberrypi:~ $ docker pull dlmcpaul/enphasecollector-arm:0.25
0.25: Pulling from dlmcpaul/enphasecollector-arm
df9b9388f04a: Already exists 
67dca4d3c9d3: Pull complete 
93f308e998d2: Pull complete 
c4dfdfe4f897: Pull complete 
0d923ba112e3: Pull complete 
Digest: sha256:4de498d1d1b8bb5b7bf6f6a3b975886bd1a2f245b9ac4eebb020b240616cf700
Status: Downloaded newer image for dlmcpaul/enphasecollector-arm:0.25
docker.io/dlmcpaul/enphasecollector-arm:0.25
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ !-2
docker run --mount source=enphase,target=/internal_db -e TZ=Australia/Sydney -e ENVOY_CONTROLLER_PASSWORD=*** -e ENVOY_CONTROLLER_HOST=*** -e ENVOY_REFRESHSECONDS=15000 -e ENVOY_PAYMENTPERKILOWATT=0.102 -e ENVOY_CHARGEPERKILOWATT=0.288 -e ENVOY_DAILYSUPPLYCHARGE=1.31 -e ENVOY_BANDS[0].FROM=0700 -e ENVOY_BANDS[0].TO=0859 -e ENVOY_BANDS[0].COLOUR=#FA8072 -e ENVOY_BANDS[1].FROM=1700 -e ENVOY_BANDS[1].TO=1959 -e ENVOY_BANDS[1].COLOUR=#FA8072 -e ENVOY_BANDS[2].FROM=0900 -e ENVOY_BANDS[2].TO=1659 -e ENVOY_BANDS[2].COLOUR=#F0E68C -e ENVOY_BANDS[3].FROM=2000 -e ENVOY_BANDS[3].TO=2159 -e ENVOY_BANDS[3].COLOUR=#F0E68C -e ENVOY_BANDS[4].FROM=2200 -e ENVOY_BANDS[4].TO=0659 -e ENVOY_BANDS[4].COLOUR=#87CEEB -p 8080:8080 dlmcpaul/enphasecollector-arm:0.25 
standard_init_linux.go:219: exec user process caused: exec format error
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ uname -m
armv7l
pi@raspberrypi:~ $

dlmcpaul, 2022-07-13:
Ok lets try this

Download the Dockerfile-builder to your raspberry pi and try this command (and look out for the "dot" at the end)

docker build -f Dockerfile-builder -t smacinto/enphasecollector-arm:local --build-arg VERSION=0.25 .

You should end up with a local image called smacinto/enphasecollector-arm:local that matches your platform.

If there are errors in fetching the base images (azul/zulu-openjdk-alpine:17) then I suspect the problem is no images available for Java 17 on raspberry pi

smacinto, 2022-07-14:
Hi David,

Unfortunately the build command failed?

pi@raspberrypi:~/Solar $ docker build -f Dockerfile-builder -t smacinto/enphasecollector-arm:local --build-arg VERSION=0.25 .
Sending build context to Docker daemon  273.9MB
Step 1/18 : FROM azul/zulu-openjdk-alpine:17 as builder
17: Pulling from azul/zulu-openjdk-alpine
df9b9388f04a: Already exists 
55f492bef161: Pull complete 
Digest: sha256:c8c75963a68c73159faf721f5c89e48637ad6c888c24dba171674659693964b7
Status: Downloaded newer image for azul/zulu-openjdk-alpine:17
 ---> 0371f5fc60b9
Step 2/18 : LABEL maintainer="dlmcpaul@gmail.com"
 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
 ---> Running in b4f6cc97b314
Removing intermediate container b4f6cc97b314
 ---> b1c668fa0707
Step 3/18 : ARG VERSION
 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
 ---> Running in 0b7153d1aece
Removing intermediate container 0b7153d1aece
 ---> 32cf34fcd87d
Step 4/18 : RUN wget -P / -O app.jar https://github.com/dlmcpaul/EnphaseCollector/releases/download/${VERSION}/enphasecollector-${VERSION}.jar
 ---> [Warning] The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm/v7) and no specific platform was requested
 ---> Running in 5003423c614c
standard_init_linux.go:219: exec user process caused: exec format error
The command '/bin/sh -c wget -P / -O app.jar https://github.com/dlmcpaul/EnphaseCollector/releases/download/${VERSION}/enphasecollector-${VERSION}.jar' returned a non-zero code: 1
pi@raspberrypi:~/Solar $

dlmcpaul, 2022-07-14:
Ok, so this confirms my suspicions.  It is not my build process failing it is a lack of a arm/v/7 java 17 image.

I will see if I can build a jar based on a lower version where there is an arm/v/7 image

dlmcpaul, 2022-07-14:
Ok there is a dlmcpaul/enphasecollector-arm:0.25 image that uses Java 11 as a base image that should have a arm/v/7 os layer

smacinto, 2022-07-15:
Thanks David - that one worked.

dlmcpaul, 2022-07-15:
Glad to hear it.  Might be a problem in future versions as I do want to start using Java 17.  Might have to figure out how to install a java 17 runtime into a arm/v/7 base OS