dlmcpaul/EnphaseCollector #7: Getting EnphaseCollector running on Raspberry Pi

solrak44, 2020-09-13:
Hello! We've just installed our solar setup and I'd love to have a permanent display running in our kitchen. Your app would be utterly perfect (it's is *way* better than what comes with the envoy!)

I got it working on my laptop but have been unsuccessful getting it to run on a Pi 4. I installed docker and docker-compose, but when I use `docker-compose up` I just get `exec user process caused "exec format error"`

Is it possible to run this on a Pi?

Thanks heaps

solrak44, 2020-09-13:
Further experimentation trying Java instead of docker, but still no joy. `mvn package` fails to build. It fails one of the tests.

dlmcpaul, 2020-09-13:
The basic docker run command should be enough to start with

docker run -it -p 8080:8080 -e ENVOY_CONTROLLER_HOST=xxx.xxx.xxx.xxx dlmcpaul/enphasecollector:latest

You can also download the prebuilt jar file 
wget https://github.com/dlmcpaul/EnphaseCollector/releases/download/0.20/enphasecollector-0.20.jar
java -jar enphasecollector-0.20.jar

dlmcpaul, 2020-09-13:
Hmm actually you will need an arm based docker base image.

Lets try for the straight java jar first

dlmcpaul, 2020-09-13:
I built an experimental arm docker image you could try

docker run -it -p 8080:8080 -e ENVOY_CONTROLLER_HOST=xxx.xxx.xxx.xxx dlmcpaul/enphasecollector:arm

solrak44, 2020-09-13:
The prebuilt jar worked perfectly, as did the arm docker image. Thanks for your help, just need to stick our old ipad to the wall and feed it the url of the pi. Brilliant!

dlmcpaul, 2020-09-13:
Excellent.  i will add the arm docker image to the release then.

Remember to externalise the database from the docker image so you can keep your history data if you upgrade the image.

closed 2020-09-26 by dlmcpaul