dlmcpaul/EnphaseCollector #9: installer password

ear9mrn, 2020-10-23:
Not an issue, just a question.

great tool. Thanks for sharing. Quick question regarding the data streams /stream/meter. This needs the installer username and the password (to access via digest/browser). How does EnphaseCollector access this stream without the password? I dug through the code but i could not find anything obvious how this is achieved and I cannot see a good place to provide this information (e.g. main/resources/application.properties).  

Thanks,

Pete.

dlmcpaul, 2020-10-23:
The answer is I don't use that URL.   I would like to but reverse engineering the password needed is not done yet.  The standard user/password I can generally calculate (and is basically free text) and if wrong is because the user changed it so they should be able to override.

The stream password very few people know what it is or how to get it although the code was described here https://thecomputerperson.wordpress.com/2016/08/28/reverse-engineering-the-enphase-installer-toolkit/

I do hope to one day implement that but I have found all the data I need without it.

ear9mrn, 2020-10-26:
I had a look at trying to extract it from the shared object file within the .apk file but hitting the limit of my knowledge/skills. It looks like you update every 5 minutes? So near real-time!

dlmcpaul, 2020-10-26:
I made a mistake in naming the environment variable.  I should have named it envoy.referesh-microseconds :-( 

So the default of envoy.refresh-seconds = 60000 is actually 60s intervals which I find reasonable.

Some users have lowered it but there is a current bug for some values.  15000 seems to work ok.  Will be fixed next release.

Also once set all the end of day summary and money calculations are based on it.  I really should record the value at the end of each day otherwise so you can change it and the history will be stay correct.

dlmcpaul, 2021-07-16:
So it looks like this project has managed to write the installer password extraction as python code https://github.com/sarnau/EnphaseEnergy

So that might make it easier to use.

dlmcpaul, 2022-04-16:
I have committed code to extract the installer password if I ever need to use it.

In the future I might look at using the stream values but for now closing this issue.

closed 2022-04-16 by dlmcpaul