dlmcpaul/EnphaseCollector #11: Support for two systems and older and newer style Enphase systems

kevinlieb, 2021-01-18:
I have two systems in my house: an older style system and a newer style.  With the newer style this seems to work fine.  With the older style I get multiple failures on startup.  

The older style system has the oval shaped powerline communication box "Envoy".

dlmcpaul, 2021-01-19:
Hmm, not sure how to support the older envoys.  I am relying on what documentation I can find on the internet for reverse engineering the API's

Maybe a quick test is what does http://<ENVOY_IP>/info.xml return?

kevinlieb, 2021-01-20:
Thanks for the quick response.  I won't be on the network where that unit is located for a few days.  But to confirm: the older systems are not supported out of the box with EnphaseCollector.

dlmcpaul, 2021-01-20:
That's correct, older models are not supported.  I built this using information I found on my Envoy Model so it will only work with models that support the API's I use.

It is also possible that newer models will change the API's and better protect them.

dlmcpaul, 2023-01-03:
I am not likely to support older model envoys.

I did some work recently for envoys missing CT's which may have helped and you are welcome to retry this.

closed 2023-01-03 by dlmcpaul