dlmcpaul/EnphaseCollector #12: Launch Failure - MacOS / Java 11

edwardSaladTongHands, 2021-03-07:
Hi thanks for publishing. The tool looks like it could be very useful. I downloaded enphasecollector-0.22.jar & latest Java 11 from adoptopenjdk.net. Launch gets as far as this message:
ERROR 38852 --- [           main] com.zaxxer.hikari.pool.HikariPool        : HikariPool-1 - Exception during pool initialization.

-and then a long stack dump - the 1st few lines are pasted below. (full trace enclosed). I’m able to reach the envoy locally & see stats. I tried running both with & without explicitly setting address/password. The 6-digit password works fine manually. I wondered if the new Java engine is battling with previous installation? I have this variable set:
`JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk-15.0.1.jdk/Contents/Home
`
Any suggestions are much appreciated.

```
org.h2.jdbc.JdbcSQLNonTransientConnectionException: Database may be already in use: null. Possible solutions: close all other connection(s); use the server mode [90020-200]
	at org.h2.message.DbException.getJdbcSQLException(DbException.java:622) ~[h2-1.4.200.jar!/:na]
	at org.h2.message.DbException.getJdbcSQLException(DbException.java:429) ~[h2-1.4.200.jar!/:na]
	at org.h2.message.DbException.get(DbException.java:194) ~[h2-1.4.200.jar!/:na]
	at org.h2.mvstore.db.MVTableEngine$Store.convertIllegalStateException(MVTableEngine.java:206) ~[h2-1.4.200.jar!/:na]

```

[javadump.log](https://github.com/dlmcpaul/EnphaseCollector/files/6096345/javadump.log)

dlmcpaul, 2021-03-09:
The message "The file is locked: nio:/Users/avi/solar_stats_db.mv.db" seems to indicate that the database file is in use by another application.

If this is the first time you are using the application then you could try deleting the above file or if this is an upgrade make sure the previous version is stopped,

I don't think your use of Java 15 is causing the problem although I have not tested with it I don't expect any issues.

edwardSaladTongHands, 2021-03-09:
Thank you kindly for your response. Yes - it was the locked database file.  I don't know why I had a stray locked stats database lying around, but trashing it enabled the program to run. (I would never have found that reason buried in the log file, thank you).  The display looks excellent.

closed 2021-03-09 by edwardSaladTongHands