dlmcpaul/EnphaseCollector #22: Change to Envoy local authentication with recent firmware update

jclendon, 2021-07-16:
Hi,

My firmware has been updated and it looks like the local page now has a new authentication set. It either requires authentication via the Enphase website or a token.

Have you seen this problem?

Thanks, James

[image]

dlmcpaul, 2021-07-16:
No my Enphase system is still running this Envoy firmware

Envoy Software Version: D5.0.34
Envoy Software Release: 2019-06-26

I assume that my application no longer works on this version?  What version do you see?

If they are allowing a token then maybe it is just a matter of updating the authentication routine.

dlmcpaul, 2021-07-16:
I will try and keep an eye on this thread https://community.enphase.com/s/question/0D53m00006ySLuRCAW/unimpressed-with-loss-of-local-api-connectivity-to-envoys and see what happens.

It looks like they intend to keep access just via a token they can revoke but have not yet built out the token creation page properly.

immesys, 2021-11-19:
@dlmcpaul I will try write up my findings so you can change your code. I initially thought that I would submit a PR, but all my code is in Go so it would be more effort to port to Java than I'd like.

The envoy now expects you to authenticate using a JWT. You can technically authenticate on the envoy home page using your enphase username/password but it's just a wrapper around the JWT. Here is my code for obtaining the JWT (in go)

```go
// The EPScraper struct just has some config data in it, it should be pretty 
// evident from the variable names what the data is
func (ep *EPSCraper) getJWT() (string, error) {
	// Error handling stripped for brevity 
	
	// We need to have an HTTP client that stores cookies, for reasons that
	// will become apparent
	jar, _ := cookiejar.New(nil)
	client := &http.Client{
		Jar: jar,
	}

	// First, login using your username and password
	client.PostForm("https://entrez.enphaseenergy.com/login",
		url.Values{"username":{ep.EnphaseUser}, "password":{ep.EnphasePassword}})
	// Response error checking omitted, but what we needed was the cookie, which is now in the jar
	
	// Second give system parameters
	resp2, _ := client.PostForm("https://entrez.enphaseenergy.com/entrez_tokens",
		url.Values{"Site":{ep.EnphaseSite}, "serialNum":{ep.EnphaseEnvoySerial}})
	// htmlquery is like an xpath library
	doc, _ := htmlquery.Parse(resp2.Body)
	textareas := htmlquery.Find(doc, "//textarea[@id=\"JWTToken\"]")
	// The first and only textarea with the JWTToken id is the JWT
	return htmlquery.InnerText(textareas[0]), nil
}
```

Now, in most systems the JWT can be used as a bearer token, so you'd expect that once you have the JWT you can drop in an `Authentication: Bearer <JWT>` header and hit the same envoy API endpoints but unfortunately that doesn't seem to be the case here. What you need to do is make a call to <envoy>/auth/check_jwt which will set a cookie if the JWT is valid and unexpired

```go
func (ep *EPSCraper) loadJWTIntoCookie(jwt string) error {
	// You need to empty the cookie jar before trying to load a new JWT in.
	// If you have an old cookie from a previous JWT auth, it gets confused and you don't end
	// up "refreshing" your auth. Then your system breaks when you hit the expiry time
	jar, err := cookiejar.New(nil)
	if err != nil {
		panic(err)
	}
	// Erase the cookies in the scraper's client bc the envoy doesn't seem to overwrite the 
	// jwt if there is one in the session already
	ep.client.Jar = jar

	req, _ := http.NewRequest("GET", fmt.Sprintf("%s/auth/check_jwt",ep.EnvoyHost), nil)
	req.Header.Set("Authorization", fmt.Sprintf("Bearer %s", jwt))
	ep.client.Do(req)
	// again, all error handling stripped. You'd normally check that response
	
	// All we needed was the cookie, which is now in ep.client.Jar
	return nil
}
```

Once you have that, you can hit the API endpoints. I'm not sure if these have changed since before or not, I've only used the new firmware. I personally use `<envoy>/production.json?details=1`

BUT. They also added mandatory https. So you have to hit the endpoints on HTPS not on HTTP. They use a self-signed cert so you will need to make sure your HTTP client is ok with ignoring cert errors (security amirite)

dlmcpaul, 2021-11-19:
@immesys Thank you for this.  When Enphase first mentioned moving to a token based auth I thought they were going to use OAuth or similar but it is looking more like they have devised their own scheme.

whisk3y, 2021-12-27:
@immesys The trouble with your method is the fact the tokens only last 12 hours, it'll work assuming the internet is up, but if the internet goes down you're still hosed. It's better than nothing until Enphase remove even that little bit of functionality to finish screwing everyone.

immesys, 2021-12-27:
@whisk3y oh, absolutely. I hate the tokens. It's a terrible idea to add a requirement for an internet connection to a product that people buy for disaster resilience. And I discovered that the installer token variants work on any envoy, bypassing any semblance of security. But we do what we can with what we are given....

gtdiehl, 2022-01-09:
@immesys I was wondering if you could take a look at my quick code in Python to see why it's failing? 

https://github.com/jesserizzo/envoy_reader/issues/78#issuecomment-1008195539

I maintain Python code for an Envoy library but don't have access to one running 7.x firmware.

whisk3y, 2022-01-09:
> @immesys I was wondering if you could take a look at my quick code in Python to see why it's failing? When I access any page on the Envoy such as check_wat or production.json I get 401 Authorization error.
> 
> [jesserizzo/envoy_reader#78 (comment)](https://github.com/jesserizzo/envoy_reader/issues/78#issuecomment-1008195539)
> 
> I maintain Python code for an Envoy library but don't have access to one running 7.x firmware.

This is the code in python to connect to a 7.x envoy: 

`"""Module to read production and consumption values from an Enphase Envoy on the local network."""
import asyncio
import logging
import re
import time
import jwt

from html.parser import HTMLParser
from json.decoder import JSONDecodeError
try: 
    from BeautifulSoup import BeautifulSoup
except ImportError:
    from bs4 import BeautifulSoup
#from envoy_utils.envoy_utils import EnvoyUtils

import httpx

class MyHTMLParser(HTMLParser):
    def handle_starttag(self, tag, attrs):
        print("Encountered a start tag:", tag)

    def handle_endtag(self, tag):
        print("Encountered an end tag :", tag)

    def handle_data(self, data):
        print("Encountered some data  :", data)

SITE_ID = "<SiteID from online portal>"
SERIAL_NUMBER = "<EnvoySN from online portal>"

ENDPOINT_URL_PRODUCTION_JSON = "http://{}/production.json"
ENDPOINT_URL_PRODUCTION_V1 = "http://{}/api/v1/production"
ENDPOINT_URL_PRODUCTION_INVERTERS = "http://{}/api/v1/production/inverters"
ENDPOINT_URL_PRODUCTION = "http://{}/production"
LOGIN_URL = "https://entrez.enphaseenergy.com/login"
TOKEN_URL = "https://entrez.enphaseenergy.com/entrez_tokens"
LOCAL_URL = "https://<enphaseAddress>/"

payload_login = {'username': '<OnlinePortal-EmailAddress>', 'password': '<OnlinePortalPassword>'}
payload_token = {'Site': SITE_ID, "serialNum": SERIAL_NUMBER}

headers = {'Content-Type': 'application/json'}

client = httpx.Client(verify=False)
token = ""
try:
    r = client.post(LOGIN_URL, data=payload_login)
    print(r.status_code)
    #print(r.text)
    r = client.post(TOKEN_URL, data=payload_token)
    print(r.status_code)
    #print(r.text)
    parsed_html = BeautifulSoup(r.text, "lxml")
    token = parsed_html.body.find('textarea').text
    print(token)
    decode = jwt.decode(token, options={"verify_signature": False}, algorithms="ES256")
    print(decode["exp"])
    
    header = {"Authorization": "Bearer " + token}
    r = client.get(LOCAL_URL + "auth/check_jwt", headers=header)
    #print(r.text)

    r = client.get(LOCAL_URL + "production.json?details=1", headers=header)
    #print(r.text)
finally:
    client.close()`

enkrypt3d, 2022-01-09:
> 

python tokentest4.py
200
200
Traceback (most recent call last):
  File "/root/tokentest4.py", line 54, in <module>
    token = parsed_html.body.find('textarea').text
AttributeError: 'NoneType' object has no attribute 'text'

Getting this error message when trying to run it.

whisk3y, 2022-01-09:
> > 
> 
> python tokentest4.py 200 200 Traceback (most recent call last): File "/root/tokentest4.py", line 54, in token = parsed_html.body.find('textarea').text AttributeError: 'NoneType' object has no attribute 'text'
> 
> Getting this error message when trying to run it.

What version of Python are you running? The code is written for Python3, print your r.text and see what's feeding into BeautifulSoup. BeautifulSoup isn't finding "textarea" in the html response from the POST. 

It's also entirely possible Enphase broke more stuff.

enkrypt3d, 2022-01-10:
I am on python 3.9.2

root@raspberrypi:~# python tokentest4.py
200
<!DOCTYPE html>
<html>
<head>
    <title>Enphase Authentication | Login page</title>
    <link rel="stylesheet" type="text/css" href="/main.css"/>
    <script src="/main.js"></script>
</head>
<body>

<section class="login-container">
    <section class="login">
        [image]
        <div>
            <p class="error">Wrong username or password</p>

        </div>

        <form class="login-form" action="/login" method="post">
            <input type="text" id="username" name="username" autofocus="autofocus" placeholder="Enlighten Username"/>
            <input type="password" id="password" name="password" placeholder="Enlighten Password"/>
            <input class="button" type="submit" value="Log in"/>
            <input type="text" id="codeChallenge" name="codeChallenge" value="" hidden=true/>
            <input type="text" id="redirectUri" name="redirectUri" value="" hidden=true/>
            <input type="text" id="client" name="client" value="" hidden=true/>
            <input type="text" id="serialNum" name="serialNum" value="" hidden=true/>
        </form>
    </section>
    <a class="powered-by" href="https://enphase.com/" target="_blank">Enphase Energy, Inc. © 2020</a>
</section>
</body>
</html>

200
Traceback (most recent call last):
  File "/root/tokentest4.py", line 54, in <module>
    token = parsed_html.body.find('textarea').text
AttributeError: 'NoneType' object has no attribute 'text'

enkrypt3d, 2022-01-11:
> What version of Python are you running? The code is written for Python3, print your r.text and see what's feeding into BeautifulSoup. BeautifulSoup isn't finding "textarea" in the html response from [...]

@whisk3y

OiI3ZDEwMDA1ZC0......snip
<!DOCTYPE html><h2>Valid token.</h2>

{"production":[{"type":"inverters","activeCount.........snip

This appears to be working now that I fixed the script !

pierre2113, 2022-01-22:
@whisk3y    

thanks for the code, with a few extra lines to read/write  (token, and expiration time to a file)
and using Home assistant  command_line platform  and Rest integration   I have everything running.   I've requested Enphase to increase my token expiration time from 1 hour to 6 months.  We'll see if they get something done.   The python code just gets the next token  if time has exceeded expiration time cutoff.

I haven't used Home assistant   envoy integration for some time  I could be wrong but I believe the URL HA collects data from is "http://envoy.ip/production.json"  that url even before token security changes has been having latency issues  since I had installed Batteries,  batteries are also connected to the envoy,  and firmware upgrade is required for battery, and battery operation is linked with the envoy reporting the data to it.

I use now "http://envoy.ip/ivp/meters/readings"  it is much faster response time.

enkrypt3d, 2022-01-22:
https://github.com/gtdiehl/core/tree/envoy_new_fw/homeassistant/components/enphase_envoy

Update your envoy integration with these files and it should be working
again. I am on the latest version of the envoy firmware and it seems to be
working well.

psylapse, 2022-02-18:
Sorry to ask a newbie question in such informed circles, but I have yet to power up my array (IQ7 + Envoy) and hence have no access to first hand data.  

What proportion of the local data presented by the Envoy as web pages, is currently accessible through the JSON process? 
I presume that we lost all possibility of actually controlling the Envoy via a local interface some years ago (for example power throttling, yes I understand that on/off is available through the installers interface)

Hopefully when I am up and running I can return the favour by contributing

dlmcpaul, 2022-02-18:
As I understand it all the data is available as JSON or XML responses from the Envoy.  But for firmware v5 and below there are 3 levels of security (none, owner, installer).

None: Most information is available without needing a login. (See https://github.com/dlmcpaul/EnphaseCollector/blob/master/src/main/java/com/hz/configuration/EnphaseRestClientConfig.java for the URL's I know)

Owner: Using the owner user id and password you can get additional information (See https://github.com/dlmcpaul/EnphaseCollector/blob/master/src/main/java/com/hz/configuration/EnphaseSecureRestClientConfig.java)

Installer: Using an installer user id and password you can get additional information and potentially update the envoy

The Installer password is used generally for the streaming interface which returns real time updates.  I find that for what I need it is unnecessary but if you want to build a reactive framework it could be useful. 

All this changes in V7 and it looks like all URL's require a token

enkrypt3d, 2022-04-15:
@psylapse What data are you looking for specifically? 

https://github.com/briancmpbll/home_assistant_custom_envoy - this shows the batteries and inverters... as mentioned before the meter/readings URL shows data updated every second.

edasque, 2022-06-23:
Since @immesys 's code was very useful to me, I thought I would share equivalent code to obtain the long term token (6 months):
```
// 6 months token: https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num=SERIALNUMBER
func getLongLivedJWT() (string, error) {

	jar, _ := cookiejar.New(nil)
	client := &http.Client{
		Jar: jar,
	}

	// First, login using your username and password
	fieldsLogin := url.Values{"user[email]": {config.String("enphase.EnphaseUser")}, "user[password]": {config.String("enphase.EnphasePassword")}}

	_, errLogin := client.PostForm("https://enlighten.enphaseenergy.com//login/login", fieldsLogin)
	// Response error checking omitted, but what we needed was the cookie, which is now in the jar

	if errLogin != nil {
		log.Fatalln(errLogin)
	}

	req, _ := http.NewRequest("GET", fmt.Sprintf("https://enlighten.enphaseenergy.com/entrez-auth-token?serial_num=%s", config.String("enphase.EnphaseEnvoySerial")), nil)
	requestResponse, requestError := client.Do(req)
	if requestError != nil {
		log.Fatalln(requestError)
	}
	if debug {
		log.Println(requestResponse)
	}

	defer requestResponse.Body.Close()

	body, err := ioutil.ReadAll(requestResponse.Body)
	if err != nil {
		log.Fatalln(err)
	}

	jwtToken := JWTToken{}
	unmarshalError := json.Unmarshal([]byte(body), &jwtToken)

	if unmarshalError != nil {
		log.Fatalln(unmarshalError)
	}
	log.Println("Retrieved long-lived JWT token successfully")
	return jwtToken.Token, nil

}
```

@dlmcpaul rather than continuing to whip up my own, I'd love to use yours, are you planning to update the code to work with the new, insecure and annoying authentication method?

dlmcpaul, 2022-06-26:
@edasque I can do so but will need a tester as my envoy is still thankfully using v5.

Most likely the first change will be to just set the token manually before moving to full automation.

edasque, 2022-07-02:
@dlmcpaul for the streaming stuff, do you hit the endpoint with BasicAuth, do you know what that auth would be for D7.0.107 (00f3a9)? I've tried admin/installer/envoy with the last 6 digits of the serial number but I can't get to it. I felt I was able to before a software update.

enkrypt3d, 2022-07-02:
You still have to authenticate with a token using your enlighten account.

edasque, 2022-07-02:
for https://envoy/stream/meter ? @enkrypt3d I tried with the JWT token as a Bearer token but that didn't seem to work either. Oddly in a browser, I can't access it either (after being logged into the local UI). 401 everywhere.

enkrypt3d, 2022-07-02:
I'm able to access it locally from my browser once logged in. It does take
about a minute to load while logging in.

dlmcpaul, 2022-07-02:
Prior to V7 the streaming endpoint needed the installer user/password and realm auth.  The installer password was generated from the serial number and was fairly straight forward.

From V7 onward it is all token based and involves parsing HTML to get the token.

I have read about issues with the streaming endpoints but I don't have v7 to test.

Not sure why your token is not working.  Maybe clear browser cache and recreate the token?

edasque, 2022-07-02:
@enkrypt3d I swear to god it worked yesterday but doesn't now. Not even through the browser after generating and entering a JWT token. And it doesn't work in my code with a Bearer header.

@dlmcpaul ready to test for you when you are.

enkrypt3d, 2022-07-02:
Try giving your envoy a reboot?

dlmcpaul, 2022-07-02:
@edasque How do you run EnphaseCollector?  Docker image or java jar?

dlmcpaul, 2022-07-03:
Can anyone who has V7 installed check if the http://<ENVOY_IP>/info.xml requires authentication or not?
if not send me the output (after redacting your serial number)

pierre2113, 2022-07-03:
I have D7.3.75, they just updated again from D7.0.71     did not need authentication.

edasque, 2022-07-05:
@dlmcpaul , I have documented the auth process here, would love your feedback: https://github.com/edasque/enphaseLocalToInfluxDB#authorization-flow

edasque, 2022-07-06:
@dlmcpaul I am ready to be your guinea pig for V7. I can run it from Java which I am guessing would be easier for you since you wouldn't have to product a test image, I can just run it from your branch. Let me know what you'd like me to do.

dlmcpaul, 2022-07-07:
Ok, I am working on adding a config variable that you can set to the token generated from the webpage then authenticate and get the cookie.  I want to test the functionality of the app first to see if there are any issues.  Then I will figure out the best way to obtain the token and keep it up to date

dlmcpaul, 2022-07-23:
@edasque I have a built an experimental docker image for testing.  it is dlmcpaul/enphasecollector:experimental

You need to get the token from the webpage first.  I want to see what works with that before coding up the website parsing code

So you can run it with

docker run -it -e TZ=Australia/Sydney -e ENVOY_CONTROLLER_HOST=<IP> -e ENVOY_BEARER_TOKEN=<TOKEN> -p 8080:8080 dlmcpaul/enphasecollector:experimental

and see what it outputs.

edasque, 2022-07-23:
On it though to be transparent, my solar is offline (blown fuse) right now, waiting for the electrician.

When I run it (on Mac M1 so arm64 running an amd64 image), I get the following and nothing more:

> 2022-07-23 10:23:16.814  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : Starting EnphaseCollectorApplication using Java 17.0.3 on 88be223e6c99 with PID 1 (/app started by root in /)
> 2022-07-23 10:23:16.832  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : No active profile set, falling back to 1 default profile: "default"

Runing the following: ```docker run -it -e TZ=America/New_York -e ENVOY_CONTROLLER_HOST=10.0.0.149 -e ENVOY_BEARER_TOKEN=eyJraWQ___LONG_6_MONTHS_JWT_Token_EC1xPHYLb_ERSTUQ -p 8080:8080 dlmcpaul/enphasecollector:experimental```

dlmcpaul, 2022-07-23:
Ok, so I have pushed another version that ignores the trust chain since it looks like the cert is self signed.

I also put an arm version under dlmcpaul/enphasecollector-arm:experimental

edasque, 2022-07-23:
Getting this now

> 2022-07-23 13:35:29.647  INFO 1 --- [           main] c.h.c.EnphaseV7RestClientConfig          : Configuring insecure RestTemplate for V7 Envoy controller endpoint http://10.0.0.149/home.json
2022-07-23 13:35:29.932 ERROR 1 --- [           main] c.h.c.EnphaseV7RestClientConfig          : Attempt to validate bearer token eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUzI1NiJ9.tokentokentokeniaGVydGVsLmRhc3F1ZUBnbWFpbC5jb20ifQ.Tcyml95xmqb0wOdnxwgGp5REVA1T8R08MNQ3OzQO0Mixd0GG8ksaQdzFwA6ORVkmo3ueaNEC1xPHYLb_ERSTUQ against http://10.0.0.149/auth/check_jwt failed with result HTTP/1.1 401 Unauthorized
2022-07-23 13:35:29.933  INFO 1 --- [           main] c.h.c.EnphaseV7RestClientConfig          : Cookie Store now has 0 cookies after result HTTP/1.1 401 Unauthorized
2022-07-23 13:35:29.941  INFO 1 --- [           main] c.h.c.EnphaseV7RestClientConfig          : Configuring protected RestTemplate for V7 Envoy controller endpoint http://10.0.0.149/api/v1/production/inverters
2022-07-23 13:35:30.164 ERROR 1 --- [           main] c.h.c.EnphaseV7RestClientConfig          : Attempt to validate bearer token eyJraWQiOiI3ZDEwMDA1ZC03ODk5LTRkMGQtYmNiNC0yNDRmOThlZTE1NmIiLCJ0eXAiOiJKV1QiLCJhbGciOiJFUtokentokentokeniaGVydGVsLmRhc3F1ZUBnbWFpbC5jb20ifQ.Tcyml95xmqb0wOdnxwgGp5REVA1T8R08MNQ3OzQO0Mixd0GG8ksaQdzFwA6ORVkmo3ueaNEC1xPHYLb_ERSTUQ against http://10.0.0.149/auth/check_jwt failed with result HTTP/1.1 401 Unauthorized
2022-07-23 13:35:30.164  INFO 1 --- [           main] c.h.c.EnphaseV7RestClientConfig          : Cookie Store now has 0 cookies after result HTTP/1.1 401 Unauthorized
2022-07-23 13:35:30.180  INFO 1 --- [           main] c.hz.components.ReleaseInfoContributor   : Application Release version is 0.25-EXP
^C2022-07-23 13:35:30.766  INFO 1 --- [           main] o.s.b.a.w.s.WelcomePageHandlerMapping    : Adding welcome page template: index

dlmcpaul, 2022-07-23:
Hmm, the token I am logging out doesn't look to be a valid JWT

Can you test it against jwt.io

edasque, 2022-07-23:
I'll double check but I bet it's valid, I just edited it for security purpose so if you're checking my logs copied here it's not valid.

dlmcpaul, 2022-07-23:
Hmm, maybe confirm that the token I log matches the token you input.  I might be reading it wrong.

dlmcpaul, 2022-07-24:
Ok new version pushed.  It looks like I forgot the Bearer text in the Authorization header

edasque, 2022-07-24:
Looks better. I think the remaining issue comes from a unhandled exception (maybe because while the gateway is online but the system is offline, not producing, I have an electrical issue, blown fuse). Though maybe not, looks like you're casting a float into an int?

(Are you looking for a job as a Spring dev btw?)

> 2022-07-24 08:26:13.234 ERROR 1 --- [   scheduling-1] com.hz.services.EnphaseService           : Failed to retrieve Solar stats. Exception was Error while extracting response for type [class com.hz.models.envoy.json.System] and content type [application/json]; nested exception is org.springframework.http.converter.HttpMessageNotReadableException: JSON parse error: Cannot deserialize value of type `int` from String "2.90": not a valid `int` value; nested exception is com.fasterxml.jackson.databind.exc.InvalidFormatException: Cannot deserialize value of type `int` from String "2.90": not a valid `int` value
 at [Source: (org.springframework.util.StreamUtils$NonClosingInputStream); line: 1, column: 91] (through reference chain: com.hz.models.envoy.json.System["db_percent_full"])

org.springframework.web.client.RestClientException: Error while extracting response for type [class com.hz.models.envoy.json.System] and content type [application/json]; nested exception is org.springframework.http.converter.HttpMessageNotReadableException: JSON parse error: Cannot deserialize value of type `int` from String "2.90": not a valid `int` value; nested exception is com.fasterxml.jackson.databind.exc.InvalidFormatException: Cannot deserialize value of type `int` from String "2.90": not a valid `int` value
 at [Source: (org.springframework.util.StreamUtils$NonClosingInputStream); line: 1, column: 91] (through reference chain: com.hz.models.envoy.json.System["db_percent_full"])
	at org.springframework.web.client.HttpMessageConverterExtractor.extractData(HttpMessageConverterExtractor.java:120) ~[spring-web-5.3.21.jar:5.3.21]
	at org.springframework.web.client.RestTemplate$ResponseEntityResponseExtractor.extractData(RestTemplate.java:1037) ~[spring-web-5.3.21.jar:5.3.21]
	at org.springframework.web.client.RestTemplate$ResponseEntityResponseExtractor.extractData(RestTemplate.java:1020) ~[spring-web-5.3.21.jar:5.3.21]
	at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:778) ~[spring-web-5.3.21.jar:5.3.21]
	at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:711) ~[spring-web-5.3.21.jar:5.3.21]
	at org.springframework.web.client.RestTemplate.getForEntity(RestTemplate.java:361) ~[spring-web-5.3.21.jar:5.3.21]
	at com.hz.services.EnphaseService.collectEnphaseData(EnphaseService.java:70) ~[app/:na]

dlmcpaul, 2022-07-24:
Ok pushed a version with db_percent_full as a float.  This is the part that worries me, they may have changed the data model for V7 so we may have lots of issues.

My main job is enterprise architect.  I do this for fun and to keep my dev skills at least a little relevant.  Trying to avoid too much ivory tower so not looking for a job (yet, maybe after the next in an endless series of restructures :-)

edasque, 2022-07-24:
Looks like that did it. On Monday hopefully, we'll have solar production again and I'll make sure the stats are good (consumption looks good thus far). I'll also get it going on a permanent host rather than my laptop and start sending data to InfluxDB or use the prometheus metrics.

edasque, 2022-07-27:
@dlmcpaul , seems to be working well. Let me know if you need me to do / test anything.

Note that it doesn't draw any excess on the graph even though I can see it in the influxdb excess metric. I am not sure how it's supposed to behave though.

[image]

dlmcpaul, 2022-07-27:
Thanks for your testing.  It is good to see it working.

Ok, so next will be the html screen scraping code which should be easy enough.  The main problem is token refresh.  With 6 month tokens it is possibly less critical to do but it would be nice.

The excess value should be 0 unless you set the envoy_export-limit variable to something other than 0 (default).  Basically this feature is for systems that are limited to a max export value (ie 10000).  if production exceeds this value then they don't get paid for the excess and I try to graph that.

gridImport and gridExport are calculated as the sum of consumption-production over time based on the collection frequency.

You should be able to calculate them using the prometheus metrics
solar.meter.import and solar.collection.period should let you calculate grid import over a period and
solar.meter.export will similarly get grid export.

If you use the default collection period of 60 seconds then it should just be sum(solar.meter.import) / 60000

paradizelost, 2022-08-30:
@dlmcpaul I've gotten up and running with the experimental version as well. 

As far as the auth process, personally, i'd rather have to update a token in a configuration file every 6 months (or be prompted for it when it expires) than give my enphase credentials.

dlmcpaul, 2022-08-30:
Yes, I have been thinking about what to deliver here.

I have written the modules for logging into the enphase site with credentials and obtaining the JWT (Thanks to @edasque for documenting it all) but still trying to understand what a good way to put it all together is (especially with the expiry + renew requirement).

But I have decided that whatever I deliver will allow for both providing a token OR providing enphase credentials.  Mainly because changing the website to stop the login is something they might do but changing the JWT auth method would need a Envoy Release.

flamalex, 2022-09-30:
> token = parsed_html.body.find('textarea').text
> AttributeError: 'NoneType' object has no attribute 'text'

hello, I too have this error, what is the "appropriate" code

dlmcpaul, 2022-09-30:
@flamalex probably not much help here.  This is for issues with my EnphaseCollector application which uses Java no Python.

I would first log out the jwt you received and check it at jwt.io to see if it parses out ok.

juetchris83, 2022-11-16:
Hi, I recently added Solar Panels with Enphase tech, and I have the same issue with : 
2022-11-16 14:25:00.948 ERROR 5172 --- [   scheduling-1] com.hz.services.EnphaseService           : Failed to retrieve Solar stats. Exception was I/O error on GET request for "http://192.168.x.x/home.json": PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target; nested exception is javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Same thing with Java or Docker. Do you have any clue ?

dlmcpaul, 2022-11-20:
@juetchris83 Just pushed a new experimental version (x86) dlmcpaul/enphasecollector:experimental

If you are running V7 firmware you can manually generate a token and pass it across using

docker run -it -e ENVOY_CONTROLLER_HOST=? -e ENVOY_BEARER_TOKEN=? -p 8080:8080 dlmcpaul/enphasecollector:experimental

replace the ? as appropriate

If you don't want to generate the token manually you can try

docker run -it -e ENVOY_CONTROLLER_HOST=? -e ENVOY_ENPHASE_WEB_USER=? -e ENVOY_ENPHASE_WEB_PASSWORD=? -p 8080:8080 dlmcpaul/enphasecollector:experimental

replace the ? as appropriate

You will need to have an account with Enphase to work with V7 firmware.

enkrypt3d, 2022-11-20:
Does enphase have lifetime tokens yet?

dlmcpaul, 2022-11-20:
No lifetime tokens.  I think the default is now 6 months.  So for the manual case above you will need to handle that.  I do display in the stats the expiry date to assist here.

If you use the second format then enphasecollector will auto renew the token.

In both scenarios you can lose access to the API if enphase disables your web login.

dlmcpaul, 2022-12-18:
Latest release should support V7 envoys.  Both auto renew and manually supplying tokens should work.

dlmcpaul, 2023-01-03:
With the latest release V7 is supported so will close this ticket.  Please open a new ticket if you have issues

closed 2023-01-03 by dlmcpaul