dlmcpaul/EnphaseCollector #29: Adding "on-peak" and "off-peak" colouring to the production/consumption graph

solrak44, 2021-12-13:
Kia ora,

I love your visualiser software. We have it proudly displayed in the kitchen and it's much remarked upon by visitors to our house! It frequently prompts us to ask "why are we using so much power?" and go looking for the offending appliance. It also prompts us to do something with the power once the hot water has finished heating. Awsome stuff!

Our power company is offering us on-peak and off-peak rates for our power at very set times of the day. I wondered how easy it would be add some background colour to the production/consumption graph to show these times (and prompt us to avoid them for high power use appliances). Is this a couple of lines or a complete re-write?

Thanks again for your awesome software, and please ignore this if it's a cheeky thing to ask :-)

Karl

dlmcpaul, 2021-12-13:
A quick question.

Do you have multiple off and on peaks or just the one on-peak.

One on peak might be easy to produce something like this
[image]

But not yet sure how to add config for it.

dlmcpaul, 2022-01-01:
@solrak44 

I have added some support for bands within the graph.  Please try out the new release (See the readme for setup).

Have not added support in the docker image though so hopefully you run the jar.

solrak44, 2022-01-09:
Sorry for the late reply - I've been away. While I was originally using the RPi, I'm running as a docker image on my new Synology. I'll see if I can get the RPi version up and running again and test this for you. It looks great!

dlmcpaul, 2022-01-09:
Ok, I am looking at how to add this feature to the Docker image so it will be there eventually

dlmcpaul, 2022-01-31:
The docker image should now also work with this feature, you will need to bind a new volume.  I have added instructions to the readme.

Let me know of any issues

solrak44, 2022-02-01:
Works perfectly reading the properties file from the NAS!

http://thinkdifferent.co.nz:8000/solar/

The only issue I have is I can’t get it to remember the data from restart to restart. I have mapped /internal_db to a folder on my NAS, but it never seems to load it back in.

Thanks so much for all you do. Your beautiful graphs and gauges never fail to intrigue people who visit our place!!

Ngā mihi nui

Karl

dlmcpaul, 2022-02-01:
Interesting, is a file being created in the folder you mapped /internal_db to?

solrak44, 2022-02-01:
Yes, there is a file called “solar_stats_db.mv.db” but it hasn’t been modified since June last year. Should I delete it and let it recreate itself?

dlmcpaul, 2022-02-01:
Not yet.  Put the docker run command you are using and let me see if there is something there.

solrak44, 2022-02-01:
So I don’t run it from the commandline – I’m running it via the docker GUI on my Synology box (DSM 7). Here is a screenshot of the running container, with the variables I selected at the start. You can see I have two folders on the NAS in a containing folder called “docker”. The properties folder contains the config file for the new version of your app. I know the app can see that folder because it correctly displays the bands I set for on-peak, offpeak (as well as the various charges and rates).

Ngā mihi nui

Karl

dlmcpaul, 2022-02-01:
Hmm, sorry no image attached,

Maybe compare the ownership information of the 2 files and the 2 directories.  It certainly looks like a lack of write permission.

solrak44, 2022-02-01:
Is the image there now?

And yes, I’ll check to make sure write permissions is correct. Thanks again for your help!

I think I owe you a beer

solrak44, 2022-02-01:
What it does do, is constantly write this to the logs now (these are the start times of the two bands I set up)

2022-02-02 11:57:46.516  INFO 1 --- [nio-8080-exec-5] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:57:46.552  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:57:46.552  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:57:54.517  INFO 1 --- [nio-8080-exec-3] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:57:54.518  INFO 1 --- [nio-8080-exec-3] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:57:59.155  INFO 1 --- [nio-8080-exec-7] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:57:59.155  INFO 1 --- [nio-8080-exec-7] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:02.520  INFO 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:02.520  INFO 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:10.501  INFO 1 --- [nio-8080-exec-2] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:10.502  INFO 1 --- [nio-8080-exec-2] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:11.531  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:11.531  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:18.501  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:18.501  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:24.555  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:24.555  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:26.510  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:26.511  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:35.500  INFO 1 --- [nio-8080-exec-7] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:35.502  INFO 1 --- [nio-8080-exec-7] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:36.841  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:36.842  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:43.569  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:43.570  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:48.828  INFO 1 --- [nio-8080-exec-5] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:48.829  INFO 1 --- [nio-8080-exec-5] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:51.524  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:51.524  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:58:59.517  INFO 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:58:59.518  INFO 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:02.260  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:02.261  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:07.851  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:07.853  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:14.697  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:14.697  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:15.574  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:15.575  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:23.522  INFO 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:23.522  INFO 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:26.763  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:26.764  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:31.525  INFO 1 --- [nio-8080-exec-2] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:31.527  INFO 1 --- [nio-8080-exec-2] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:39.580  INFO 1 --- [nio-8080-exec-5] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:39.580  INFO 1 --- [nio-8080-exec-5] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:39.943  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:39.944  INFO 1 --- [nio-8080-exec-8] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:47.382  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 0700
2022-02-02 11:59:47.382  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 1700
2022-02-02 11:59:52.238  INFO 1 --- [nio-8080-exec-5] com.hz.controllers.EnphaseController     : 0700
[... cut ...]
2022-02-02 12:00:27.530  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 0700
2022-02-02 12:00:27.530  INFO 1 --- [nio-8080-exec-6] com.hz.controllers.EnphaseController     : 1700
2022-02-02 12:00:30.772  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 0700
2022-02-02 12:00:30.773  INFO 1 --- [io-8080-exec-10] com.hz.controllers.EnphaseController     : 1700
2022-02-02 12:00:35.540  INFO 1 --- [nio-8080-exec-4] com.hz.controllers.EnphaseController     : 0700

dlmcpaul, 2022-02-01:
Yes sorry, that's a debug I left in there.

Try sending any images to my email dlmcpaul AT gmail.com

solrak44, 2022-02-01:
Yep, I’m not stressed about it, just though you might need to know :-)

I’ll try the other email

dlmcpaul, 2022-02-05:
Ok, I have added a default for the database to the docker file.  Try the latest image.  It should also remove the logging.

solrak44, 2022-02-08:
Have updated the container, and it’s working perfectly.

Thanks heaps!

dlmcpaul, 2022-02-08:
Cool.  I will close this issue off then.

closed 2022-02-08 by dlmcpaul