dlmcpaul/EnphaseCollector #30: Log4j vulnerability CVE-2021-44228

ErikLentz, 2021-12-14:
I noticed this project imports log4j. Would it be impacted by CVE-2021-44228?

dlmcpaul, 2021-12-14:
I am using the default logging package provided by spring boot and haven't included the affected log4j2 package.

So based on this https://spring.io/blog/2021/12/10/log4j2-vulnerability-and-spring-boot I am not affected by the CVE.

This project is also wired into the Snyk vulnerability scanner and it has not raised this as an issue.

dlmcpaul, 2021-12-14:
Thank you for keeping an eye out for such things.

I expect once I finish work for the year and head into Christmas I should find some time to release a new version.  This will include upgrades to the latest jars.

closed 2021-12-14 by dlmcpaul