dlmcpaul/EnphaseCollector #37: Duplicate key error on new deployment when writing to influxdb

paradizelost, 2022-08-30:
on latest version of influx i get: 
enphaseCollector    | 2022-08-30 00:21:34.814  INFO 1 --- [           main] com.hz.configuration.InfluxDBConfig      : Writing solar stats to influx database at http://db:8086
enphaseCollector    | 2022-08-30 00:21:35.093  WARN 1 --- [           main] ConfigServletWebServerApplicationContext : Exception encountered during context initialization - cancelling refresh attempt: org.springframework.beans.factory.UnsatisfiedDependencyException: Error creating bean with name 'influxService' defined in file [/app/com/hz/services/InfluxService.class]: Unsatisfied dependency expressed through constructor parameter 0; nested exception is org.springframework.beans.factory.BeanCreationException: Error creating bean with name 'destinationInfluxDB' defined in class path resource [com/hz/configuration/InfluxDBConfig.class]: Bean instantiation via factory method failed; nested exception is org.springframework.beans.BeanInstantiationException: Failed to instantiate [org.influxdb.InfluxDB]: Factory method 'destinationInfluxDB' threw exception; nested exception is org.influxdb.InfluxDBException: {"code":"unauthorized","message":"unauthorized access"}
enphaseCollector    | 2022-08-30 00:21:35.096  INFO 1 --- [           main] j.LocalContainerEntityManagerFactoryBean : Closing JPA EntityManagerFactory for persistence unit 'default'
enphaseCollector    | 2022-08-30 00:21:35.103  INFO 1 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Shutdown initiated...
enphaseCollector    | 2022-08-30 00:21:35.143  INFO 1 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Shutdown completed.
enphaseCollector    | 2022-08-30 00:21:35.212 ERROR 1 --- [           main] i.micrometer.influx.InfluxMeterRegistry  : unable to create database 'collectorStats': {"code":"unauthorized","message":"unauthorized access"}
enphaseCollector    | 2022-08-30 00:21:35.293 ERROR 1 --- [           main] i.micrometer.influx.InfluxMeterRegistry  : failed to send metrics to influx: {"code":"unauthorized","message":"unauthorized access"}
enphaseCollector    | 2022-08-30 00:21:35.310 ERROR 1 --- [           main] i.micrometer.influx.InfluxMeterRegistry  : unable to create database 'collectorStats': {"code":"unauthorized","message":"unauthorized access"}
enphaseCollector    | 2022-08-30 00:21:35.337 ERROR 1 --- [           main] i.micrometer.influx.InfluxMeterRegistry  : failed to send metrics to influx: {"code":"unauthorized","message":"unauthorized access"}
enphaseCollector    | 2022-08-30 00:21:35.340  INFO 1 --- [           main] o.apache.catalina.core.StandardService   : Stopping service [Tomcat]
enphaseCollector    | 2022-08-30 00:21:35.385  INFO 1 --- [           main] ConditionEvaluationReportLoggingListener :

Trying to use influx 1.8,  Error i'm receiving is:

`enphaseCollector    | 2022-08-30 00:30:21.515  WARN 1 --- [nio-8080-exec-9] o.h.engine.jdbc.spi.SqlExceptionHelper   : SQL Error: 23505, SQLState: 23505
enphaseCollector    | 2022-08-30 00:30:21.516 ERROR 1 --- [nio-8080-exec-9] o.h.engine.jdbc.spi.SqlExceptionHelper   : Unique index or primary key violation: "PUBLIC.UK_ESIJSR1QSXCG3GCQ4AF42OF09_INDEX_5 ON PUBLIC.EVENT_PANELS(PANELS_ID) VALUES 41"; SQL statement:
enphaseCollector    | insert into event_panels (event_id, panels_id) values (?, ?) [23505-200]
enphaseCollector    | 2022-08-30 00:30:21.517  INFO 1 --- [nio-8080-exec-9] o.h.e.j.b.internal.AbstractBatchImpl     : HHH000010: On release of batch it still contained JDBC statements
enphaseCollector    | 2022-08-30 00:30:21.525 ERROR 1 --- [nio-8080-exec-9] com.hz.controllers.EnphaseController     : populateMultiStatsStatusList Exception: could not execute statement; SQL [n/a]; constraint ["PUBLIC.UK_ESIJSR1QSXCG3GCQ4AF42OF09_INDEX_5 ON PUBLIC.EVENT_PANELS(PANELS_ID) VALUES 41"; SQL statement:
enphaseCollector    | insert into event_panels (event_id, panels_id) values (?, ?) [23505-200]]; nested exception is org.hibernate.exception.ConstraintViolationException: could not execute statement
enphaseCollector    |
enphaseCollector    | org.springframework.dao.DataIntegrityViolationException: could not execute statement; SQL [n/a]; constraint ["PUBLIC.UK_ESIJSR1QSXCG3GCQ4AF42OF09_INDEX_5 ON PUBLIC.EVENT_PANELS(PANELS_ID) VALUES 41"; SQL statement:
enphaseCollector    | insert into event_panels (event_id, panels_id) values (?, ?) [23505-200]]; nested exception is org.hibernate.exception.ConstraintViolationException: could not execute statement
enphaseCollector    |   at org.springframework.orm.jpa.vendor.HibernateJpaDialect.convertHibernateAccessException(HibernateJpaDialect.java:276) ~[spring-orm-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.orm.jpa.vendor.HibernateJpaDialect.translateExceptionIfPossible(HibernateJpaDialect.java:233) ~[spring-orm-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.orm.jpa.JpaTransactionManager.doCommit(JpaTransactionManager.java:566) ~[spring-orm-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.transaction.support.AbstractPlatformTransactionManager.processCommit(AbstractPlatformTransactionManager.java:743) ~[spring-tx-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.transaction.support.AbstractPlatformTransactionManager.commit(AbstractPlatformTransactionManager.java:711) ~[spring-tx-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.transaction.interceptor.TransactionAspectSupport.commitTransactionAfterReturning(TransactionAspectSupport.java:654) ~[spring-tx-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.transaction.interceptor.TransactionAspectSupport.invokeWithinTransaction(TransactionAspectSupport.java:407) ~[spring-tx-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.transaction.interceptor.TransactionInterceptor.invoke(TransactionInterceptor.java:119) ~[spring-tx-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.aop.framework.ReflectiveMethodInvocation.proceed(ReflectiveMethodInvocation.java:186) ~[spring-aop-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.aop.framework.CglibAopProxy$CglibMethodInvocation.proceed(CglibAopProxy.java:763) ~[spring-aop-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.aop.framework.CglibAopProxy$DynamicAdvisedInterceptor.intercept(CglibAopProxy.java:708) ~[spring-aop-5.3.21.jar:5.3.21]
enphaseCollector    |   at com.hz.services.LocalDBService$$EnhancerBySpringCGLIB$$be5d2602.getMaxPanelProduction(<generated>) ~[app/:na]
enphaseCollector    |   at com.hz.controllers.EnphaseController.populateMultiStatsStatusList(EnphaseController.java:80) ~[app/:na]
enphaseCollector    |   at com.hz.controllers.EnphaseController.populateStatusList(EnphaseController.java:100) ~[app/:na]
enphaseCollector    |   at com.hz.controllers.EnphaseController.status(EnphaseController.java:149) ~[app/:na]
enphaseCollector    |   at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[na:na]
enphaseCollector    |   at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77) ~[na:na]
enphaseCollector    |   at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:na]
enphaseCollector    |   at java.base/java.lang.reflect.Method.invoke(Method.java:568) ~[na:na]
enphaseCollector    |   at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:205) ~[spring-web-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:150) ~[spring-web-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:117) ~[spring-webmvc-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:895) ~[spring-webmvc-5.3.21.jar:5.3.21]
enphaseCollector    |   at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:808) ~[spring-webmvc-5.3.21.jar:5.3.21]
[... cut ...]
enphaseCollector    |   at com.zaxxer.hikari.pool.ProxyPreparedStatement.executeUpdate(ProxyPreparedStatement.java:61) ~[HikariCP-4.0.3.jar:na]
enphaseCollector    |   at com.zaxxer.hikari.pool.HikariProxyPreparedStatement.executeUpdate(HikariProxyPreparedStatement.java) ~[HikariCP-4.0.3.jar:na]
enphaseCollector    |   at org.hibernate.engine.jdbc.internal.ResultSetReturnImpl.executeUpdate(ResultSetReturnImpl.java:197) ~[hibernate-core-5.6.9.Final.jar:5.6.9.Final]
enphaseCollector    |   ... 92 common frames omitted
enphaseCollector    |`

dlmcpaul, 2022-08-30:
My quick look suggests that the influx db is protected by a user/password?

dlmcpaul, 2023-01-03:
Ok, 2 different errors here:

The first error is because I currently use an influxdb library that only supports up to version 1.8 only.

The second I may have fixed by marking certain transactions as read only in a recent commit but not sure if it fixed as the error is very rare.  I need to look again at removing hibernate so I can control better the commits.

dlmcpaul, 2023-01-04:
#42 Might fix this

dlmcpaul, 2023-03-07:
@paradizelost if you are able to build from source the current changes are expected to fix these issues.

Note: if you have an existing database it will need to be upgraded to 2.0

dlmcpaul, 2023-04-30:
Please check with latest version and re-open if still an issue

closed 2023-04-30 by dlmcpaul