dlmcpaul/EnphaseCollector #40: Receiving error after updating to current version

paradizelost, 2022-12-05:
Caused by: org.springframework.http.converter.HttpMessageNotReadableException: JSON parse error: Cannot deserialize value of type `int` from String "0.36": not a valid Integer value; nested exception is com.fasterxml.jackson.databind.exc.InvalidFormatException: Cannot deserialize value of type `int` from String "0.36": not a valid Integer value

dlmcpaul, 2022-12-05:
Do you run as a java jar or the docker image?

If the docker image can you try the dlmcpaul/enphasecollector:experimental image

paradizelost, 2022-12-05:
So, i'm getting java SSL errors on this version. Looks like it doesn't like the self-signed ssl certificate that the envoy uses: 

`enphaseCollector    | Caused by: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
enphaseCollector    |   at java.base/sun.security.ssl.Alert.createSSLException(Alert.java:131) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:371) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:314) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.TransportContext.fatal(TransportContext.java:309) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.checkServerCerts(CertificateMessage.java:654) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.onCertificate(CertificateMessage.java:473) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.consume(CertificateMessage.java:369) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.SSLHandshake.consume(SSLHandshake.java:396) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:480) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:458) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.TransportContext.dispatch(TransportContext.java:201) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.SSLTransport.decode(SSLTransport.java:172) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1505) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1420) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:455) ~[na:na]
enphaseCollector    |   at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:426) ~[na:na]
enphaseCollector    |   at org.apache.http.conn.ssl.SSLConnectionSocketFactory.createLayeredSocket(SSLConnectionSocketFactory.java:436) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.conn.ssl.SSLConnectionSocketFactory.connectSocket(SSLConnectionSocketFactory.java:384) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.conn.DefaultHttpClientConnectionOperator.connect(DefaultHttpClientConnectionOperator.java:142) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.conn.PoolingHttpClientConnectionManager.connect(PoolingHttpClientConnectionManager.java:376) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.execchain.MainClientExec.establishRoute(MainClientExec.java:393) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.execchain.MainClientExec.execute(MainClientExec.java:236) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.execchain.ProtocolExec.execute(ProtocolExec.java:186) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.execchain.RetryExec.execute(RetryExec.java:89) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.execchain.RedirectExec.execute(RedirectExec.java:110) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.client.InternalHttpClient.doExecute(InternalHttpClient.java:185) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:83) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.apache.http.impl.client.CloseableHttpClient.execute(CloseableHttpClient.java:56) ~[httpclient-4.5.13.jar:4.5.13]
enphaseCollector    |   at org.springframework.http.client.HttpComponentsClientHttpRequest.executeInternal(HttpComponentsClientHttpRequest.java:87) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.AbstractBufferingClientHttpRequest.executeInternal(AbstractBufferingClientHttpRequest.java:48) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.AbstractClientHttpRequest.execute(AbstractClientHttpRequest.java:66) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.BufferingClientHttpRequestWrapper.executeInternal(BufferingClientHttpRequestWrapper.java:63) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.AbstractBufferingClientHttpRequest.executeInternal(AbstractBufferingClientHttpRequest.java:48) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.AbstractClientHttpRequest.execute(AbstractClientHttpRequest.java:66) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.InterceptingClientHttpRequest$InterceptingRequestExecution.execute(InterceptingClientHttpRequest.java:109) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.boot.actuate.metrics.web.client.MetricsClientHttpRequestInterceptor.intercept(MetricsClientHttpRequestInterceptor.java:86) ~[spring-boot-actuator-2.6.13. [... cut ...]
enphaseCollector    |   at org.springframework.http.client.AbstractBufferingClientHttpRequest.executeInternal(AbstractBufferingClientHttpRequest.java:48) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.http.client.AbstractClientHttpRequest.execute(AbstractClientHttpRequest.java:66) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:776) ~[spring-web-5.3.23.jar:5.3.23]
enphaseCollector    |   ... 17 common frames omitted`

dlmcpaul, 2022-12-05:
So was this the experimental docker image?  or the 0.25 version?

paradizelost, 2022-12-06:
Experimental
Daniel Hamik

dlmcpaul, 2022-12-06:
Ok pushed a new experimental image with some additional logging.

Are you providing a token you generated or the enphase web user/password?

dlmcpaul, 2022-12-06:
I thought I was ignoring the SSL certificates.  The relevant code 

					.setSSLContext(new SSLContextBuilder().loadTrustMaterial(new TrustAllStrategy()).build())
					.setSSLHostnameVerifier(new NoopHostnameVerifier())

from https://github.com/dlmcpaul/EnphaseCollector/blob/master/src/main/java/com/hz/services/EnvoyConnectionProxy.java should trust the cert anyway and not verify the hostname vs cert.

dlmcpaul, 2022-12-06:
Can you provide the full log up to the error and the command you are using to start the container?

I can only think that one of the non ssl calls now needs ssl.

paradizelost, 2022-12-13:
This is the ssl error i get on startup. 
[image]

if i put it behind an nginx proxy with a valid ssl certificate i get:
[image]

dlmcpaul, 2022-12-13:
Ok, I was afraid of that. Looks like all except the info.xml are now https only.

Under V5 only certain endpoints were protected.  My previous V7 testing showed them as still open but now I guess they are secured, at least behind TLS.  

Ok, I have made a new experimental build that use the protected client for all endpoints.  Please try that one.

paradizelost, 2022-12-13:
Updated and can confirm that has taken care of the issue.  Thanks!

dlmcpaul, 2022-12-13:
Ok I will likely do a full release soon