dlmcpaul/EnphaseCollector #44: Consumption values wrong when another hybrid inverter is present in the system

goldserve, 2023-01-08:
I'm not sure if some math is being preformed but when there is another inverter in the system and enphase is only part of the PV production, when I am exporting power back to grid, the consumption is the total of the export + enphase PV generation which makes consumption way too high. I don't know how to work around this but maybe an option just to show Net Grid import/export instead of trying to show consumption.

dlmcpaul, 2023-01-08:
If your system is < v7 could you send me the output of

curl http://<envoy-ip>/home.json
curl http://<envoy-ip>/inventory.json?deleted=1
curl http://<envoy-ip>production.json?details=1
curl http://<envoy-ip>/info.xml

replacing envoy-ip as appropriate

dlmcpaul, 2023-01-08:
Possibly good to get both when exporting power and not exporting power outputs

goldserve, 2023-01-08:
1) home.json

{"software_build_epoch":1607591706,"is_nonvoy":false,"db_size":"3 MB","db_percent_full":"0","timezone":"US/Pacific","current_date":"01/08/2023","current_time":"12:51","network":{"web_comm":true,"ever_reported_to_enlighten":true,"last_enlighten_report_time":1673210650,"primary_interface":"wlan0","interfaces":[{"type":"ethernet","interface":"eth0","mac":"00:1D:C0:B0:4F:0D","dhcp":true,"ip":"169.254.120.1","signal_strength":0,"signal_strength_max":1,"carrier":false},{"signal_strength":4,"signal_strength_max":5,"type":"wifi","interface":"wlan0","mac":"B0:B1:13:7F:8C:F0","dhcp":true,"ip":"192.168.10.117","carrier":true,"supported":true,"present":true,"configured":true,"status":"connected"}]},"tariff":"none","comm":{"num":16,"level":5,"pcu":{"num":16,"level":5},"acb":{"num":0,"level":0},"nsrb":{"num":0,"level":0}},"alerts":[],"update_status":"not-satisfied"}

2) inventory

[{"type":"PCU","devices":[{"part_num":"800-00625-r02","installed":"1672080043","serial_num":"202126086069","device_status":["envoy.global.ok"],"last_rpt_date":"1673210702","admin_state":1,"dev_type":1,"created_date":"1672080043","img_load_date":"1583340412","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00142-r01-v04.18.01","chaneid":1627390225,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672080052","serial_num":"202126086070","device_status":["envoy.global.ok"],"last_rpt_date":"1673210704","admin_state":1,"dev_type":1,"created_date":"1672080052","img_load_date":"1583340412","img_pnum_running":"520-00082-r01-v04.18.02","ptpn":"540-00142-r01-v04.18.01","chaneid":1627390481,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672261375","serial_num":"202126086899","device_status":["envoy.global.ok"],"last_rpt_date":"1673210707","admin_state":1,"dev_type":1,"created_date":"1672261375","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627390737,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672261384","serial_num":"202126087006","device_status":["envoy.global.ok"],"last_rpt_date":"1673210708","admin_state":1,"dev_type":1,"created_date":"1672261384","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627390993,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672273052","serial_num":"202126013634","device_status":["envoy.global.ok"],"last_rpt_date":"1673210710","admin_state":1,"dev_type":1,"created_date":"1672273052","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627391249,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672278753","serial_num":"202126084764","device_status":["envoy.global.ok"],"last_rpt_date":"1673210713","admin_state":2,"dev_type":1,"created_date":"1672278753","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627391505,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672278756","serial_num":"202126085949","device_status":["envoy.global.ok"],"last_rpt_date":"1673210714","admin_state":2,"dev_type":1,"created_date":"1672278756","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627391761,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672278757","serial_num":"202126086329","device_status":["envoy.global.ok"],"last_rpt_date":"1673210716","admin_state":2,"dev_type":1,"created_date":"1672278757","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627392017,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672278758","serial_num":"202126047231","device_status":["envoy.global.ok"],"last_rpt_date":"1673210719","admin_state":2,"dev_type":1,"created_date":"1672278758","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627392273,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672278759","serial_num":"202126086883","device_status":["envoy.global.ok"],"last_rpt_date":"1673210719","admin_state":2,"dev_type":1,"created_date":"1672278759","img_load_date":"1673110844","img_pnum_running":"520-00082-r01-v04.28.07","ptpn":"540-00142-r01-v04.28.03","chaneid":1627392529,"device_control":[{"gficlearset":false}],"producing":true,"communicating":true,"provisioned":true,"operating":true},{"part_num":"800-00625-r02","installed":"1672279040","serial_num":"202126084833","device_status":["envoy.globa [... cut ...] orage":[{"type":"acb","activeCount":0,"readingTime":0,"wNow":0,"whNow":0,"state":"idle"}]}

4)

1673211285 202114068541 800-00655-r09 D5.0.55 4c8675 0 1 true 500-00001-r01 02.00.00950 500-00011-r01 04.01.15 1905ae 590-00019-r01 02.00.01 1f421b 500-00002-r01 05.00.554f2662 500-00004-r01 01.02.293 9cf065 500-00008-r01 02.01.22 2faa48 500-00010-r0105.00.02 4fe435 500-00013-r01 03.02.07 276642 500-00012-r01 02.02.00 8969f6 500-00016-r01 02.00.00 54a6dc 1607601617 release-5.0.x-106-Dec-10-20-02:30:12

dlmcpaul, 2023-01-08:
Thanks I will see if that helps me understand what is happening.

Be aware that it may be something special in how your installer wired things up

dlmcpaul, 2023-01-09:
Ok, so both your net consumption and total consumption measuring devices in the envoy are returning negative consumption values.

For the point in time that the data was captured you had approx
total production=1288W
total consumption=-266W

Not sure how to interpret that.  The basic calculations I do is read production and consumption then export is when production > consumption and import is when consumption > production.

consumption should not go negative.

You mention you have 2 systems connected up.  Since I cannot read the other system then I doubt I can create accurate stats.

I have a few choices here, I could zero out negative consumption or treat negative as positive. maybe but I really don't know why consumption is negative.

goldserve, 2023-01-09:
So I installed the system myself and i'm the installer. I made sure to check the CTs and it is readying correctly on L1 and L2. The hybrid inverter I have with batteries also has solar so it can be generating solar and pushing back out onto the grid.

I understand it is hard to figure out what the house consumption is because there is another system at play here so maybe an option for just display Gird Import/Export is fine. It will just tell me how much power I am importing or exporting at any given time is sufficient

Cheers,
Kevin

dlmcpaul, 2023-01-09:
Unfortunately, the enphase controller doesn't report grid import or export that I know of.  It reports production and consumption values and then I calculate import/export.

The code is based on reverse engineering the data I see.  If you know of a field that supplies grid import/export then I can look at that.

Maybe in your scenario a negative consumption is grid export, does that seem right?

goldserve, 2023-01-09:
There is a measurementType of total-consumption and net-consumption. I will read this value again but I am exporting solar on both enphase and hybrid inverter array and see which value correlates with my other inverter which also has a CT at the same location and it can show import or export wattage but doesn't have an API to pull this data unfortunately.

Cheers,
Kevin

goldserve, 2023-01-09:
I also see realtime readings from this open URL:

http://envoy.local/ivp/meters/readings

apparentEnergy hilighted below seems to be the real-time power infor. Again, I will check again when there is PV production.

[
    {
        "eid": 704643328,
        "timestamp": 1673246459,
        "actEnergyDlvd": 51296.894,
        "actEnergyRcvd": 0.000,
        "apparentEnergy": 135375.794,
        "reactEnergyLagg": 84150.883,
        "reactEnergyLead": 10359.777,
        "instantaneousDemand": -0.000,
        "activePower": -0.000,
        "apparentPower": 344.493,
        "reactivePower": 339.688,
        "pwrFactor": 0.000,
        "voltage": 245.774,
        "current": 2.805,
        "freq": 60.000,
        "channels": [
            {
                "eid": 1778385169,
                "timestamp": 1673246459,
                "actEnergyDlvd": 25608.854,
                "actEnergyRcvd": 0.000,
                "apparentEnergy": 67585.658,
                "reactEnergyLagg": 42049.076,
                "reactEnergyLead": 5225.845,
                "instantaneousDemand": -0.000,
                "activePower": -0.000,
                "apparentPower": 171.924,
                "reactivePower": 169.707,
                "pwrFactor": 0.000,
                "voltage": 122.726,
                "current": 1.402,
                "freq": 60.000
            },
            {
                "eid": 1778385170,
                "timestamp": 1673246459,
                "actEnergyDlvd": 25688.040,
                "actEnergyRcvd": 0.000,
                "apparentEnergy": 67790.136,
                "reactEnergyLagg": 42101.807,
                "reactEnergyLead": 5133.932,
                "instantaneousDemand": 0.000,
                "activePower": 0.000,
                "apparentPower": 172.569,
                "reactivePower": 169.981,
                "pwrFactor": 0.000,
                "voltage": 123.048,
                "current": 1.403,
                "freq": 60.000
            },
            {
                "eid": 1778385171,
                "timestamp": 1673246459,
                "actEnergyDlvd": 0.000,
                "actEnergyRcvd": 0.000,
                "apparentEnergy": 0.000,
                "reactEnergyLagg": 0.000,
                "reactEnergyLead": 0.000,
                "instantaneousDemand": 0.000,
                "activePower": 0.000,
                "apparentPower": 0.000,
                "reactivePower": 0.000,
                "pwrFactor": 0.000,
                "voltage": 0.000,
                "current": 0.000,
                "freq": 60.000
            }
        ]
    },
    {
        "eid": 704643584,
        "timestamp": 1673246459,
        "actEnergyDlvd": 47136.082,
        "actEnergyRcvd": 12329.314,
        "apparentEnergy": 74436.960,
        "reactEnergyLagg": 2001.790,
        "reactEnergyLead": 30845.887,
        "instantaneousDemand": 1331.855,
        "activePower": 1331.855,
        "apparentPower": 1552.065,
        "reactivePower": -709.217,
        "pwrFactor": 0.856,
        "voltage": 245.858,
        "current": 12.630,
        "freq": 60.000,
        "channels": [
            {
                "eid": 1778385425,
                "timestamp": 1673246459,
                "actEnergyDlvd": 25188.413,
                "actEnergyRcvd": 5057.257,
                "apparentEnergy": 37770.792,
                "reactEnergyLagg": 582.673,
                "reactEnergyLead": 16132.798,
                "instantaneousDemand": 1003.636,
                "activePower": 1003.636,
                "apparentPower": 1142.436,
                "reactivePower": -488.030,
                "pwrFactor": 0.876,
                "voltage": 122.802,
                "current": 9.284,
                "freq": 60.000
            },
            {
                "eid": 1778385426,
                "timestamp": 1673246459,
                "actEnergyDlvd": 21947.669,
                "actEnergyRcvd": 7272.057,
                "apparentEnergy": 36666.168,
                "reactEnergyLagg": 1419.117,
                "reactEnergyLead": 14713.089,
                "instantaneousDemand": 328.220,
                "activePower": 328.220,
                "apparentPower": 409.629,
                "reactivePower": -221.187,
                "pwrFactor": 0.805,
                "voltage": 123.056,
                "current": 3.346,
                "freq": 60.000
            },
            {
                "eid": 1778385427,
                "timestamp": 1673246459,
                "actEnergyDlvd": 0.000,
                "actEnergyRcvd": 0.000,
                "apparentEnergy": 0.000,
                "reactEnergyLagg": 0.000,
                "reactEnergyLead": 0.000,
                "instantaneousDemand": 0.000,
                "activePower": 0.000,
                "apparentPower": 0.000,
                "reactivePower": 0.000,
                "pwrFactor": 0.000,
                "voltage": 0.000,
                "current": 0.000,
                "freq": 60.000
            }
        ]
    }
]

dlmcpaul, 2023-01-09:
The net consumption value is also negative.  It is also the sum of production and total consumption if both were negative.

total production=1288W
total consumption=-266W
net consumption = -1554

The endpoints under the ivp path are supposed to be realtime but require a different password so I never used them as most of the data I needed was under the open endpoints and the other data was protected by a basic scheme.

From V7 the ivp endpoints may be broken, there was some discussion on the forums about it but I cannot check.

goldserve, 2023-01-09:
> Sorry, no images are shown. Might need to link to them or email direct to dlmcpaul AT gmail.com
>
> So what is the ask here.
>
>   *   Treat negative consumption as grid export?
>   *   Have a mode to use ivr stats for consumption instead of net or total consumption?
>
> The second looks cleaner if it's possible.

Here is the picture attached.

It looks like "measurementType":"net-consumption", "wNow":-835.975 will give Import if positive and export if negative straight up, no calculations required to figure out "consumption".

dlmcpaul, 2023-01-09:
How do you run the application?  Java jar or docker image?

Can you see the images in your comments because I don't.  > 10Mb perhaps?

goldserve, 2023-01-09:
Let's see if you can see this image now?

I run the application via docker image but i'm pretty handy if you need me to make changes in the code to try out.
[cid:ffdcce8c-bb5f-4d98-b95c-4de10bb71bca]

dlmcpaul, 2023-01-11:
Ok I have a first cut of something to try.  Will build a docker image for you to test with soon.

dlmcpaul, 2023-01-14:
There is an experimental docker image that tries to generate the stats while expecting consumption to track import/export

You will need something like

docker run -it -e TZ=US/Pacific -e ENVOY_CONTROLLER_HOST=XXX -e SPRING_PROFILES_ACTIVE=experimental -p 8080:8080 dlmcpaul/enphasecollector:experimental

I have pushed the code I am using under MetricCalculatorNegativeConsumption and that is activated by the experimental profile.

Might need to find a better name for the profile if this goes ahead.

goldserve, 2023-01-18:
Cool, I just started a new docker instance with this image and will report back when I have good sunshine from both arrays. Cheers!

dlmcpaul, 2023-04-15:
@goldserve Did the new profile module work for you