dlmcpaul/EnphaseCollector #50: Automatically select https (port 443) for V7 enphase devices

dlmcpaul, 2023-03-14:
It would be good to auto select HTTPS for enphase devices running the V7 firmware but allow downgrade if required.

Might be difficult since we don't know the version until we fetch the info.xml file.

Maybe try with passed in values and if failure retry with HTTPS?

dlmcpaul, 2023-05-21:
That's good to know.  I have a change for this but not yet ready to test with.

Interestingly your version is higher than any I have seen so far.

Matthew1471, 2023-08-17:
The latest is 7.3.130 for North America and 7.6.175 for Europe.
Source: https://www.cisa.gov/news-events/ics-advisories/icsa-23-171-01

I'm working with Enphase to get a new version to fix some further vulnerabilities I discovered, so there will be another version soon. Mine is going to be the guinea pig before they seek for wider release.

RE: Port 80 vs 443, the gateway's NGINX instance will redirect all http to https unless they come from the IQ gateway itself (localhost, as some of the internal APIs consume other APIs) or in the 2 specific exceptions below :

The always HTTP allowed paths are:
- "/info" for backwards compatibility and to work out if this is a V7 token Gateway or not.
- "/admin/lib/dba" for support tunnel.

And HTTP is also allowed for when the user is accessing these URLs in access point mode:
- "/admin/lib/network_display" 
- and "/admin/lib/wireless_display"

Hope this helps,
Matthew

dlmcpaul, 2023-08-19:
> The HTTP always allowed paths are:
>
>    "/info" for backwards compatibility to work out if this is a V7 token Gateway or not.

That is good.  I am relying on being able to access /info for autoselecting/detecting v7.

closed 2023-09-06 by dlmcpaul