dlmcpaul/EnphaseCollector #53: Enphase Gateway (Envoy) API changes

ErikLentz, 2023-04-14:
[API Documentation can be found here](https://enphase.com/download/accessing-iq-gateway-local-apis-or-local-ui-token-based-authentication?utm_source=US_Installers_IQ+Gateway+Local+API+Security_In+Batches&utm_medium=email&utm_content=151115)

Here is a summary of the changes that will go into effect with release 07.03.120 and higher:
--
* Added a new capability to generate and authenticate secure access  tokens via web UI to secure all custom applications and API calls.   
* Documentation now includes examples of how to use URLs to get  tokens programmatically using shell script-based or Python-based  methods.   
* Revised documentation also explains how to connect securely using the updated IQ Gateway local UI and/or IQ Gateway APIs.

daniel981, 2023-04-15:
I just got the email today from Enphase regarding the push of the new firmware with update API and documentation. Good to know that this was already posted here. Thank you dlmcpaul, really appreciate your work on this project. 

Here is the full email for reference: 

At Enphase, we take security seriously. We want to ensure that all customers and stakeholders have access to the most secure and reliable operating environment possible.
 
We’ll be updating the API security protocols associated with the software running on the IQ Gateway, and we’re writing to share information about these changes with all Enphase homeowners, installers, software developers, and partners who may be affected.
 
These updates have begun propagating across accounts and will continue to roll out over time to all accounts. If you’re creating, using, or maintaining custom monitoring software that relies on interactions with IQ Gateway local interfaces, formally known as Envoy, this critical information will require your review and potential action.
 
Here is a summary of the changes that will go into effect with release 07.03.120 and higher:

-  Added a new capability to generate and authenticate secure access tokens via web UI to secure all custom applications and API calls.
- Documentation now includes examples of how to use URLs to get tokens programmatically using shell script-based or Python-based methods.
- Revised documentation also explains how to connect securely using the updated IQ Gateway local UI and/or IQ Gateway APIs.

Note that if your system is a beta system, an IQ8 system, or an IQ Battery-based system, you may already be running version 07.03.120 or higher on your Gateway.
 
Complete and newly updated API documentation is [available here](https://enphase.com/download/accessing-iq-gateway-local-apis-or-local-ui-token-based-authentication).
 
If you have further questions or require additional support after consulting the updated API documentation, please [reach out for support](https://support.enphase.com/s/contact-us).
 
We’re here to help, and the small investment of time and resources required to enhance these security protocols will be well worth it to safeguard the health and uninterrupted function of your Enphase systems.
 
Thank you,
 
The Enphase Team

dlmcpaul, 2023-04-15:
Hmm a quick read suggests that not much has changed.  They look to have added a more direct code friendly URL for authenticating to their website.

drache42, 2023-04-23:
using `ENVOY_ENPHASEWEBUSER` and `ENVOY_ENPHASEWEBPASSWORD` does not work for me. I get the following error.

using `ENVOY_BEARERTOKEN` does work. And it seems that once the container has it saved, then it doesn't matter. I started container with `ENVOY_BEARERTOKEN`, then stopped it. Replaced it with user/pass and it still works.

```
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:35.512  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : Starting EnphaseCollectorApplication using Java 17.0.6 on enphase with PID 1 (/app started by root in /)
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:35.515  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : No active profile set, falling back to 1 default profile: "default"
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.116  INFO 1 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Bootstrapping Spring Data JPA repositories in DEFAULT mode.
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.161  INFO 1 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Finished Spring Data repository scanning in 39 ms. Found 5 JPA repository interfaces.
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.625  INFO 1 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat initialized with port(s): 8080 (http)
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.634  INFO 1 --- [           main] o.apache.catalina.core.StandardService   : Starting service [Tomcat]
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.635  INFO 1 --- [           main] org.apache.catalina.core.StandardEngine  : Starting Servlet engine: [Apache Tomcat/9.0.68]
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.734  INFO 1 --- [           main] o.a.c.c.C.[Tomcat].[localhost].[/solar]  : Initializing Spring embedded WebApplicationContext
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:36.734  INFO 1 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 1181 ms
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.023  INFO 1 --- [           main] o.hibernate.jpa.internal.util.LogHelper  : HHH000204: Processing PersistenceUnitInfo [name: default]
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.073  INFO 1 --- [           main] org.hibernate.Version                    : HHH000412: Hibernate ORM core version 5.6.12.Final
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.074  INFO 1 --- [           main] org.hibernate.cfg.Environment            : HHH000205: Loaded properties from resource hibernate.properties: {hibernate.bytecode.use_reflection_optimizer=false, hibernate.bytecode.provider=none}
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.118  INFO 1 --- [           main] o.hibernate.annotations.common.Version   : HCANN000001: Hibernate Commons Annotations {5.1.2.Final}
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.207  INFO 1 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Starting...
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.360  INFO 1 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Start completed.
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.385  INFO 1 --- [           main] org.hibernate.dialect.Dialect            : HHH000400: Using dialect: org.hibernate.dialect.H2Dialect
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.854  INFO 1 --- [           main] o.h.e.t.j.p.i.JtaPlatformInitiator       : HHH000490: Using JtaPlatform implementation: [org.hibernate.engine.transaction.jta.platform.internal.NoJtaPlatform]
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.860  INFO 1 --- [           main] j.LocalContainerEntityManagerFactoryBean : Initialized JPA EntityManagerFactory for persistence unit 'default'
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:37.917  INFO 1 --- [           main] c.h.c.EnphaseSystemInfoConfig            : Reading system information from Envoy controller endpoint http://192.168.107.15/info.xml
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:38.263  INFO 1 --- [           main] c.hz.components.ReleaseInfoContributor   : Application Release version is 0.26
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:38.859  INFO 1 --- [           main] o.s.b.a.w.s.WelcomePageHandlerMapping    : Adding welcome page template: index
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:39.641  INFO 1 --- [           main] o.s.b.a.e.web.EndpointLinksResolver      : Exposing 6 endpoint(s) beneath base path '/actuator'
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:39.690  INFO 1 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat started on port(s): 8080 (http) with context path '/solar'
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:39.708  INFO 1 --- [   scheduling-1] com.hz.utils.EnphaseJWTExtractor         : Fetching Enlighten Login Page
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:39.710  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : Started EnphaseCollectorApplication in 4.453 seconds (JVM running for 4.727)
enphasecollector-enphasecollector-1  | 2023-04-23 06:56:39.948  INFO 1 --- [           main] com.hz.services.LocalDBService           : Summarising Event table prior to 2023-04-23T00:00
[... cut ...]
enphasecollector-enphasecollector-1  |  at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) ~[na:na]
enphasecollector-enphasecollector-1  |  at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[na:na]
enphasecollector-enphasecollector-1  |  at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[na:na]
enphasecollector-enphasecollector-1  |  at java.base/java.lang.Thread.run(Thread.java:833) ~[na:na]
```

dlmcpaul, 2023-04-23:
You look to be running release 0.26 instead of from the current code base but I can see that the current code base is failing to log into the enphase website.  Thinks it logged in but got a null token.  Please try building from source and running that.

com.hz.utils.EnphaseJWTExtractor         : Attempting to Login with a Form Submit
com.hz.utils.EnphaseJWTExtractor         : SubmitForm Status = HTTP/1.1 302 Found with redirect to https://enlighten.enphaseenergy.com/web/4037105?v=3.4.0
com.hz.utils.EnphaseJWTExtractor         : Fetching and Scanning returned HTML for Token
com.hz.services.EnvoyConnectionProxy     : Attempt to validate bearer token **null** against

dlmcpaul, 2023-04-30:
Please test against the 0.27 release.

rice1204, 2023-05-04:
Appreciate this great project!

I'd like to try it out, but I'm having trouble connecting the docker container to my V7 envoy. It seems to be having trouble with authentication (either with username/password or with ENVOY_BEARER_TOKEN). 

Have I overlooked something?

```
> docker run -it -e ENVOY_CONTROLLER_HOST=192.168.5.10 -e ENVOY_ENPHASE_WEB_USER=xxxxxxx@xxxxx.com -e ENVOY_ENPHASE_WEB_PASSWORD=xxxxxx dlmcpaul/enphasecollector

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v3.0.6)

2023-05-04T19:54:54.114Z  INFO 7 --- [           main] com.hz.EnphaseCollectorApplication       : Starting EnphaseCollectorApplication using Java 17.0.7 with PID 7 (/app started by root in /)
2023-05-04T19:54:54.118Z  INFO 7 --- [           main] com.hz.EnphaseCollectorApplication       : No active profile set, falling back to 1 default profile: "default"
2023-05-04T19:54:54.993Z  INFO 7 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Bootstrapping Spring Data JPA repositories in DEFAULT mode.
2023-05-04T19:54:55.057Z  INFO 7 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Finished Spring Data repository scanning in 56 ms. Found 5 JPA repository interfaces.
2023-05-04T19:54:55.738Z  WARN 7 --- [           main] io.undertow.websockets.jsr               : UT026010: Buffer pool was not set on WebSocketDeploymentInfo, the default pool will be used
2023-05-04T19:54:55.756Z  INFO 7 --- [           main] io.undertow.servlet                      : Initializing Spring embedded WebApplicationContext
2023-05-04T19:54:55.758Z  INFO 7 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 1432 ms
2023-05-04T19:54:55.960Z  INFO 7 --- [           main] o.f.c.internal.license.VersionPrinter    : Flyway Community Edition 9.5.1 by Redgate
2023-05-04T19:54:55.960Z  INFO 7 --- [           main] o.f.c.internal.license.VersionPrinter    : See what's new here: https://flywaydb.org/documentation/learnmore/releaseNotes#9.5.1
2023-05-04T19:54:55.960Z  INFO 7 --- [           main] o.f.c.internal.license.VersionPrinter    :
2023-05-04T19:54:55.967Z  INFO 7 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Starting...
2023-05-04T19:54:56.133Z  INFO 7 --- [           main] com.zaxxer.hikari.pool.HikariPool        : HikariPool-1 - Added connection conn0: url=jdbc:h2:/internal_db/solar_stats_db_v2 user=SA
2023-05-04T19:54:56.135Z  INFO 7 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Start completed.
2023-05-04T19:54:56.145Z  INFO 7 --- [           main] o.f.c.i.database.base.BaseDatabaseType   : Database: jdbc:h2:/internal_db/solar_stats_db_v2 (H2 2.1)
2023-05-04T19:54:56.190Z  INFO 7 --- [           main] o.f.core.internal.command.DbValidate     : Successfully validated 2 migrations (execution time 00:00.009s)
2023-05-04T19:54:56.196Z  INFO 7 --- [           main] o.f.c.i.s.JdbcTableSchemaHistory         : Creating Schema History table "PUBLIC"."flyway_schema_history" ...
2023-05-04T19:54:56.215Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Current version of schema "PUBLIC": << Empty Schema >>
2023-05-04T19:54:56.225Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Migrating schema "PUBLIC" to version "1 - InitSchema"
2023-05-04T19:54:56.241Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Migrating schema "PUBLIC" to version "2 - MigrateH2DatabaseToV2"
2023-05-04T19:54:56.245Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Successfully applied 2 migrations to schema "PUBLIC", now at version v2 (execution time 00:00.031s)
2023-05-04T19:54:56.304Z  INFO 7 --- [           main] o.hibernate.jpa.internal.util.LogHelper  : HHH000204: Processing PersistenceUnitInfo [name: default]
2023-05-04T19:54:56.340Z  INFO 7 --- [           main] org.hibernate.Version                    : HHH000412: Hibernate ORM core version 6.1.7.Final
2023-05-04T19:54:56.342Z  INFO 7 --- [           main] org.hibernate.cfg.Environment            : HHH000205: Loaded properties from resource hibernate.properties: {hibernate.type.preferred_instant_jdbc_type=TIMESTAMP, hibernate.bytecode.use_reflection_optimizer=false, hibernate.id.db_structure_naming_strategy=single, hibernate.bytecode.provider=none, hibernate.type.preferred_duration_jdbc_type=BIGINT}
2023-05-04T19:54:56.479Z  INFO 7 --- [           main] SQL dialect                              : HHH000400: Using dialect: org.hibernate.dialect.H2Dialect
2023-05-04T19:54:56.499Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.type.preferred_duration_jdbc_type].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
2023-05-04T19:54:56.500Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.type.preferred_instant_jdbc_type].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
[... cut ...]
        at org.springframework.web.client.RestTemplate.createResourceAccessException(RestTemplate.java:888) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:868) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:764) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.getForObject(RestTemplate.java:378) ~[spring-web-6.0.8.jar:6.0.8]
        at com.hz.services.EnvoyService.getProductionData(EnvoyService.java:130) ~[app/:na]
```

drache42, 2023-05-04:
> docker run -it -e ENVOY_CONTROLLER_HOST=192.168.5.10 -e ***@***.*** -e ENVOY_ENPHASE_WEB_PASSWORD=xxxxxx dlmcpaul/enphasecollector

I've had a similar issue. I had to use a token instead of username/password.

-Steffen

rice1204, 2023-05-04:
Thanks for the tip. I'm getting similar results with token auth too. Is it something wrong with my docker command?

**docker run -it -e ENVOY_CONTROLLER_HOST=192.168.5.10 -e ENVOY_BEARER_TOKEN=[super_long_token] dlmcpaul/enphasecollector**

```
Failed to retrieve Solar stats. Exception was I/O error on GET request for "http://192.168.5.10/production.json": Socket closed
```

```

  .   ____          _            __ _ _
 /\\ / ___'_ __ _ _(_)_ __  __ _ \ \ \ \
( ( )\___ | '_ | '_| | '_ \/ _` | \ \ \ \
 \\/  ___)| |_)| | | | | || (_| |  ) ) ) )
  '  |____| .__|_| |_|_| |_\__, | / / / /
 =========|_|==============|___/=/_/_/_/
 :: Spring Boot ::                (v3.0.6)

2023-05-04T22:09:59.038Z  INFO 7 --- [           main] com.hz.EnphaseCollectorApplication       : Starting EnphaseCollectorApplication using Java 17.0.7 with PID 7 (/app started by root in /)
2023-05-04T22:09:59.043Z  INFO 7 --- [           main] com.hz.EnphaseCollectorApplication       : No active profile set, falling back to 1 default profile: "default"
2023-05-04T22:09:59.981Z  INFO 7 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Bootstrapping Spring Data JPA repositories in DEFAULT mode.
2023-05-04T22:10:00.062Z  INFO 7 --- [           main] .s.d.r.c.RepositoryConfigurationDelegate : Finished Spring Data repository scanning in 73 ms. Found 5 JPA repository interfaces.
2023-05-04T22:10:00.799Z  WARN 7 --- [           main] io.undertow.websockets.jsr               : UT026010: Buffer pool was not set on WebSocketDeploymentInfo, the default pool will be used
2023-05-04T22:10:00.818Z  INFO 7 --- [           main] io.undertow.servlet                      : Initializing Spring embedded WebApplicationContext
2023-05-04T22:10:00.820Z  INFO 7 --- [           main] w.s.c.ServletWebServerApplicationContext : Root WebApplicationContext: initialization completed in 1576 ms
2023-05-04T22:10:01.047Z  INFO 7 --- [           main] o.f.c.internal.license.VersionPrinter    : Flyway Community Edition 9.5.1 by Redgate
2023-05-04T22:10:01.047Z  INFO 7 --- [           main] o.f.c.internal.license.VersionPrinter    : See what's new here: https://flywaydb.org/documentation/learnmore/releaseNotes#9.5.1
2023-05-04T22:10:01.047Z  INFO 7 --- [           main] o.f.c.internal.license.VersionPrinter    :
2023-05-04T22:10:01.056Z  INFO 7 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Starting...
2023-05-04T22:10:01.253Z  INFO 7 --- [           main] com.zaxxer.hikari.pool.HikariPool        : HikariPool-1 - Added connection conn0: url=jdbc:h2:/internal_db/solar_stats_db_v2 user=SA
2023-05-04T22:10:01.255Z  INFO 7 --- [           main] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Start completed.
2023-05-04T22:10:01.268Z  INFO 7 --- [           main] o.f.c.i.database.base.BaseDatabaseType   : Database: jdbc:h2:/internal_db/solar_stats_db_v2 (H2 2.1)
2023-05-04T22:10:01.318Z  INFO 7 --- [           main] o.f.core.internal.command.DbValidate     : Successfully validated 2 migrations (execution time 00:00.009s)
2023-05-04T22:10:01.324Z  INFO 7 --- [           main] o.f.c.i.s.JdbcTableSchemaHistory         : Creating Schema History table "PUBLIC"."flyway_schema_history" ...
2023-05-04T22:10:01.347Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Current version of schema "PUBLIC": << Empty Schema >>
2023-05-04T22:10:01.357Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Migrating schema "PUBLIC" to version "1 - InitSchema"
2023-05-04T22:10:01.377Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Migrating schema "PUBLIC" to version "2 - MigrateH2DatabaseToV2"
2023-05-04T22:10:01.382Z  INFO 7 --- [           main] o.f.core.internal.command.DbMigrate      : Successfully applied 2 migrations to schema "PUBLIC", now at version v2 (execution time 00:00.037s)
2023-05-04T22:10:01.444Z  INFO 7 --- [           main] o.hibernate.jpa.internal.util.LogHelper  : HHH000204: Processing PersistenceUnitInfo [name: default]
2023-05-04T22:10:01.486Z  INFO 7 --- [           main] org.hibernate.Version                    : HHH000412: Hibernate ORM core version 6.1.7.Final
2023-05-04T22:10:01.494Z  INFO 7 --- [           main] org.hibernate.cfg.Environment            : HHH000205: Loaded properties from resource hibernate.properties: {hibernate.type.preferred_instant_jdbc_type=TIMESTAMP, hibernate.bytecode.use_reflection_optimizer=false, hibernate.id.db_structure_naming_strategy=single, hibernate.bytecode.provider=none, hibernate.type.preferred_duration_jdbc_type=BIGINT}
2023-05-04T22:10:01.643Z  INFO 7 --- [           main] SQL dialect                              : HHH000400: Using dialect: org.hibernate.dialect.H2Dialect
2023-05-04T22:10:01.665Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.type.preferred_duration_jdbc_type].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
2023-05-04T22:10:01.668Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.type.preferred_instant_jdbc_type].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
[... cut ...]
2023-05-04T22:10:12.744Z  INFO 7 --- [           main] org.jboss.threads                        : JBoss Threads version 3.5.0.Final
2023-05-04T22:10:12.788Z  INFO 7 --- [           main] o.s.b.w.e.undertow.UndertowWebServer     : Undertow started on port(s) 8080 (http) with context path '/solar'
2023-05-04T22:10:12.845Z  INFO 7 --- [           main] com.hz.EnphaseCollectorApplication       : Started EnphaseCollectorApplication in 14.182 seconds (process running for 14.706)
2023-05-04T22:10:12.991Z  INFO 7 --- [           main] com.hz.services.LocalDBService           : Summarising Event table prior to 2023-05-04T00:00
```

drache42, 2023-05-04:
This log doesn't have the error, so it looks like it's working.

That or you didn't wait long so the error hasn't show up. That "Summarising
Event table prior to" step can take a while, so be patient.

-Steffen

dlmcpaul, 2023-05-04:
Hmm, with the switch to V7 firmware I think you need SSL when connecting to the envoy.

Can you try adding ENVOY_CONTROLLER_PORT=443

rice1204, 2023-05-05:
Thanks, but sadly, still no luck. At least the docker image is connecting now with https, so I'm getting closer!
Looks like authentication still the hangup for me. I bet it's something really simple I'm overlooking.

Just to reconfirm, the ENVOY_BEARER_TOKEN is supposed to be the one manually retrieved from https://entrez.enphaseenergy.com/entrez_tokens right?

```
docker run -it -e ENVOY_CONTROLLER_PORT=443 -e ENVOY_CONTROLLER_HOST=192.168.5.10 -e ENVOY_BEARER_TOKEN=[super long string] -p 8080:8080 dlmcpaul/enphasecollector
```

```
2023-05-05T01:15:54.651Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.id.db_structure_naming_strategy].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
2023-05-05T01:15:54.661Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.type.preferred_duration_jdbc_type].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
2023-05-05T01:15:54.661Z  WARN 7 --- [           main] org.hibernate.orm.incubating             : HHH90006001: Encountered incubating setting [hibernate.type.preferred_instant_jdbc_type].  See javadoc on corresponding `org.hibernate.cfg.AvailableSettings` constant for details.
2023-05-05T01:15:55.020Z  INFO 7 --- [           main] o.h.e.t.j.p.i.JtaPlatformInitiator       : HHH000490: Using JtaPlatform implementation: [org.hibernate.engine.transaction.jta.platform.internal.NoJtaPlatform]
2023-05-05T01:15:55.027Z  INFO 7 --- [           main] j.LocalContainerEntityManagerFactoryBean : Initialized JPA EntityManagerFactory for persistence unit 'default'
2023-05-05T01:15:55.536Z  INFO 7 --- [           main] c.h.c.EnphaseSystemInfoConfig            : Reading system information from Envoy controller endpoint https://192.168.5.10/info.xml
2023-05-05T01:15:56.089Z  INFO 7 --- [           main] com.hz.utils.MetricCalculatorStandard    : Standard Consumption Metric Calculator Activated
2023-05-05T01:15:56.175Z  INFO 7 --- [           main] io.undertow                              : starting server: Undertow - 2.3.5.Final
2023-05-05T01:15:56.181Z  INFO 7 --- [           main] org.xnio                                 : XNIO version 3.8.8.Final
2023-05-05T01:15:56.189Z  INFO 7 --- [           main] org.xnio.nio                             : XNIO NIO Implementation Version 3.8.8.Final
2023-05-05T01:15:56.246Z  INFO 7 --- [           main] org.jboss.threads                        : JBoss Threads version 3.5.0.Final
2023-05-05T01:15:56.285Z  INFO 7 --- [           main] o.s.b.w.e.undertow.UndertowWebServer     : Undertow started on port(s) 8080 (http) with context path '/solar'
2023-05-05T01:15:56.327Z  INFO 7 --- [           main] com.hz.EnphaseCollectorApplication       : Started EnphaseCollectorApplication in 4.577 seconds (process running for 5.097)
2023-05-05T01:15:56.464Z  INFO 7 --- [           main] com.hz.services.LocalDBService           : Summarising Event table prior to 2023-05-05T00:00
^C2023-05-05T01:16:57.816Z  INFO 7 --- [ionShutdownHook] io.undertow                              : stopping server: Undertow - 2.3.5.Final
2023-05-05T01:16:57.822Z  INFO 7 --- [ionShutdownHook] j.LocalContainerEntityManagerFactoryBean : Closing JPA EntityManagerFactory for persistence unit 'default'
2023-05-05T01:16:57.825Z  INFO 7 --- [ionShutdownHook] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Shutdown initiated...
2023-05-05T01:16:57.827Z  INFO 7 --- [ionShutdownHook] com.zaxxer.hikari.HikariDataSource       : HikariPool-1 - Shutdown completed.
2023-05-05T01:16:57.823Z ERROR 7 --- [   scheduling-1] com.hz.services.EnvoyService             : Failed to retrieve Solar stats. Exception was I/O error on GET request for "https://192.168.5.10/home.json": Socket closed

org.springframework.web.client.ResourceAccessException: I/O error on GET request for "https://192.168.5.10/home.json": Socket closed
        at org.springframework.web.client.RestTemplate.createResourceAccessException(RestTemplate.java:888) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:868) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:764) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.getForEntity(RestTemplate.java:405) ~[spring-web-6.0.8.jar:6.0.8]
        at com.hz.services.EnvoyService.getSystemData(EnvoyService.java:46) ~[app/:na]
        at com.hz.services.EnvoyService.collectEnphaseData(EnvoyService.java:57) ~[app/:na]
        at com.hz.services.OutputManager.gather(OutputManager.java:47) ~[app/:na]
        at jdk.internal.reflect.GeneratedMethodAccessor35.invoke(Unknown Source) ~[na:na]
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:na]
        at java.base/java.lang.reflect.Method.invoke(Method.java:568) ~[na:na]
        at org.springframework.scheduling.support.ScheduledMethodRunnable.run(ScheduledMethodRunnable.java:84) ~[spring-context-6.0.8.jar:6.0.8]
        at org.springframework.scheduling.support.DelegatingErrorHandlingRunnable.run(DelegatingErrorHandlingRunnable.java:54) ~[spring-context-6.0.8.jar:6.0.8]
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) ~[na:na]
[... cut ...]
        at org.springframework.http.client.BufferingClientHttpRequestWrapper.executeInternal(BufferingClientHttpRequestWrapper.java:56) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.http.client.AbstractBufferingClientHttpRequest.executeInternal(AbstractBufferingClientHttpRequest.java:48) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.http.client.AbstractClientHttpRequest.execute(AbstractClientHttpRequest.java:66) ~[spring-web-6.0.8.jar:6.0.8]
        at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:862) ~[spring-web-6.0.8.jar:6.0.8]
        ... 16 common frames omitted

```

drache42, 2023-05-05:
Let's make sure the givens are given.

Can you go to https://192.168.5.10/home.json?
It should say not authorized, and bring you to the sign in page. Sign in
with either method and then go to https://192.168.5.10/home.json again. It
should work the second time

rice1204, 2023-05-05:
Yup, works with manual sign-in :)

home.json
`
{"software_build_epoch":1665679533,"is_nonvoy":false,"db_size":13847552,"db_percent_full":"  3.67","timezone":"US/Pacific","current_date":"05/04/2023","current_time":"19:23","network":{"web_comm":true,"ever_reported_to_enlighten":true,"last_enlighten_report_time":1683253187,"primary_interface":"wlan0","interfaces":[{"type":"ethernet","interface":"eth0","mac":"00:1D:C0:73:63:36","dhcp":true,"ip":"169.254.120.1","signal_strength":0,"signal_strength_max":1,"carrier":false},{"signal_strength":2,"signal_strength_max":5,"type":"wifi","interface":"wlan0","mac":"2C:AB:33:2A:E6:84","dhcp":true,"ip":"192.168.5.10","carrier":true,"supported":true,"present":true,"configured":true,"status":"connected"}]},"tariff":"single_rate","comm":{"num":2,"level":5,"pcu":{"num":2,"level":5},"acb":{"num":0,"level":0},"nsrb":{"num":0,"level":0},"esub":{"num":0,"level":0},"encharge":[{"num":0,"level":0,"level_24g":0,"level_subg":0}]},"alerts":[],"update_status":"not-satisfied","wireless_connection":[{"signal_strength":0,"signal_strength_max":0,"type":"zigbee","connected":false},{"signal_strength":0,"signal_strength_max":0,"type":"subghz","connected":false}],"enpower":{"connected":false,"grid_status":"closed"}}
`

dlmcpaul, 2023-05-05:
Well there are no errors in the logs about acquiring the token and if the token is invalid you should get an authorisation error not a socket error.

Is the code working for anybody else?

drache42, 2023-05-05:
I'm running latest dockerimage and it's working for me.

dlmcpaul, 2023-05-05:
Ok so maybe it is just a version thing.  Let's see what info.xml shows as the version and maybe we need some tweaks depending on what version of 7 is being run.

rice1204, 2023-05-05:
```
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<envoy_info>
<time>1683259762</time>
<device>
<sn>122035061711</sn>
<pn>800-00555-r03</pn>
<software>D7.3.120</software>
<euaid>4c8675</euaid>
<seqnum>0</seqnum>
<apiver>1</apiver>
<imeter>true</imeter>
</device>
<web-tokens>true</web-tokens>
<package name="rootfs">
<pn>500-00001-r01</pn>
<version>02.00.00</version>
<build>1210</build>
</package>
<package name="kernel">
<pn>500-00011-r02</pn>
<version>04.04.225</version>
<build>3eb4d3</build>
</package>
<package name="boot">
<pn>590-00018-r01</pn>
<version>02.00.01</version>
<build>426697</build>
</package>
<package name="app">
<pn>500-00002-r01</pn>
<version>07.03.120</version>
<build>ac5b3c</build>
</package>
<package name="devimg">
<pn>500-00004-r01</pn>
<version>01.02.413</version>
<build>9c64cd</build>
</package>
<package name="geo">
<pn>500-00008-r01</pn>
<version>02.01.24</version>
<build>a74d96</build>
</package>
<package name="backbone">
<pn>500-00010-r01</pn>
<version>07.00.20</version>
<build>176d57</build>
</package>
<package name="meter">
<pn>500-00013-r01</pn>
<version>03.02.08</version>
<build>f9ead3</build>
</package>
<package name="agf">
<pn>500-00012-r01</pn>
<version>02.02.00</version>
<build>887e04</build>
</package>
<package name="security">
<pn>500-00016-r01</pn>
<version>02.00.00</version>
<build>54a6dc</build>
</package>
<package name="essimg">
<pn>500-00020-r01</pn>
<version>22.10.09</version>
<build>983d9b</build>
</package>
<package name="pkgsec">
<pn>500-00021-r01</pn>
<version>01.00.00</version>
<build>19ae14</build>
</package>
<build_info>
<build_id>ec2-user-envoy_uber-pkg_master:pkg-Oct-13-22-17:48:05</build_id>
<build_time_gmt>1665683434</build_time_gmt>
<release_ver>02.00.3244</release_ver>
<release_stage>700-GA</release_stage>
</build_info>
</envoy_info>
```

dlmcpaul, 2023-05-05:
So D7.3.120 is not working

@drache42 what version you on?

rice1204, 2023-05-05:
pardon me, but for some reason, I tried it again today and it's working now with the manual token! Thanks!
(I ran it bare metal in JAVA on the same machine and it worked. Afterward, I tried the docker image again and it was also fine. I still haven't been able to get it running correctly with email/password auth.)

I have a question about the parameters for getting mqtt running, but I'll start a separate thread for that.

ChadDa3mon, 2023-05-18:
Wanted to chime in here and first say a big thank you for your work on this. 

I'm on on version D7.3.123 and having issues with the container as well. I tried adding `ENVOY_CONTROLLER_PORT=443` to my docker-compose, but now I'm getting errors I think are related to the self-signed certificate (I'm not a java guy, sorry). 

```
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:439) ~[na:na]
	at java.base/sun.security.validator.PKIXValidator.engineValidate(PKIXValidator.java:306) ~[na:na]
	at java.base/sun.security.validator.Validator.validate(Validator.java:264) ~[na:na]
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkTrusted(X509TrustManagerImpl.java:231) ~[na:na]
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(X509TrustManagerImpl.java:132) ~[na:na]
	at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.checkServerCerts(CertificateMessage.java:638) ~[na:na]
	... 57 common frames omitted
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.provider.certpath.SunCertPathBuilder.build(SunCertPathBuilder.java:146) ~[na:na]
	at java.base/sun.security.provider.certpath.SunCertPathBuilder.engineBuild(SunCertPathBuilder.java:127) ~[na:na]
	at java.base/java.security.cert.CertPathBuilder.build(CertPathBuilder.java:297) ~[na:na]
	at java.base/sun.security.validator.PKIXValidator.doBuild(PKIXValidator.java:434) ~[na:na]
	... 62 common frames omitted
```

I appear to be on docker container version 0.27

```
# cat release.properties 
release.version=0.27
```

A bit from info.xml (Let me know if you need more)
```
<envoy_info>
	<time>1684426458</time>
	<device>
		<sn>123456</sn>
		<pn>800-00647-r10</pn>
		<software>D7.3.123</software>
		<euaid>XYZ</euaid>
		<seqnum>0</seqnum>
		<apiver>1</apiver>
		<imeter>true</imeter>
	</device>
	<web-tokens>true</web-tokens>
```

dlmcpaul, 2023-05-21:
I will need the full set of error logs.  Java is nice enough to give detailed errors but it tends to hide the error location.

rkokfhung, 2023-07-10:
Hi @dlmcpaul , just wondering if you could help. Been using your project for the last two years - works great so thanks very much for this.

My envoy has recently updated so things stopped working. I tried to generate a token through https://entrez.enphaseenergy.com/ but I get a page with Select System / Select Gateway after logging in. Both these fields are blanked and there's nothing in the drop down list.  I am guessing that's something wrong with my Enphase account? I  can login to https://enlighten.enphaseenergy.com/web/2112623?v=3.4.0 using same credential and see my system though.

Abandoning the token method, I tried the username/password method. But still couldn't authenticate. Any suggestions how I could generate a token or get the username/password method working? Many thanks.

2023-07-10 16:21:12.096  INFO 1 --- [           main] c.hz.components.ReleaseInfoContributor   : Application Release version is 0.26                             
2023-07-10 16:21:14.282  INFO 1 --- [           main] o.s.b.a.w.s.WelcomePageHandlerMapping    : Adding welcome page template: index                             
2023-07-10 16:21:16.851  INFO 1 --- [           main] o.s.b.a.e.web.EndpointLinksResolver      : Exposing 6 endpoint(s) beneath base path '/actuator'            
2023-07-10 16:21:16.989  INFO 1 --- [           main] o.s.b.w.embedded.tomcat.TomcatWebServer  : Tomcat started on port(s): 8080 (http) with context path '/solar
'                                                                                                                                                                
2023-07-10 16:21:17.065  INFO 1 --- [           main] com.hz.EnphaseCollectorApplication       : Started EnphaseCollectorApplication in 22.52 seconds (JVM runnin
g for 23.552)                                                                                                                                                    
2023-07-10 16:21:17.072  INFO 1 --- [   scheduling-1] com.hz.utils.EnphaseJWTExtractor         : Fetching Enlighten Login Page                                   
2023-07-10 16:21:17.830  INFO 1 --- [           main] com.hz.services.LocalDBService           : Summarising Event table prior to 2023-07-10T00:00               
2023-07-10 16:21:18.830  INFO 1 --- [   scheduling-1] com.hz.utils.EnphaseJWTExtractor         : Attempting to Login with a Form Submit                          
2023-07-10 16:21:21.720  INFO 1 --- [   scheduling-1] com.hz.utils.EnphaseJWTExtractor         : SubmitForm Status = HTTP/1.1 302 Found with redirect to https://
enlighten.enphaseenergy.com/web/2112623?v=3.4.0                                                                                                                  
2023-07-10 16:21:21.724  INFO 1 --- [   scheduling-1] com.hz.utils.EnphaseJWTExtractor         : Fetching and Scanning returned HTML for Token                   
2023-07-10 16:21:23.193 ERROR 1 --- [   scheduling-1] com.hz.services.EnvoyConnectionProxy     : Attempt to validate bearer token null against http://192.168.1.9
2/auth/check_jwt failed with result HTTP/1.1 401 Unauthorized                                                                                                    
2023-07-10 16:21:25.536 ERROR 1 --- [   scheduling-1] com.hz.services.EnvoyService             : Failed to retrieve Solar stats. Exception was 401 Unauthorized: 
"<html><EOL><EOL><head><title>401 Authorization Required</title></head><EOL><EOL><body><EOL><EOL><center><h1>401 Authorization Required</h1></center><EOL><EOL><h
r><center>Redirecting to <a id='link' href=''></a></center><EOL><EOL><script><EOL><EOL>const redirect_url = window.location.protocol + '//' + window.location.hos
t + '/home';<EOL><EOL>function redirect() { window.location.href = redirect_url; }<EOL><EOL>window.onload = function() {<EOL><EOL>    const link = document.getEl
ementById('link');<EOL><EOL>    link.href = redirect_url;<EOL><EOL>    link.text = redirect_url;<EOL><EOL>    setTimeout(redirect, 500);<EOL><EOL>}<EOL><EOL></sc
ript><EOL><EOL></body><EOL><EOL></html><EOL>"                                                                                                                    
                                                                                                                                                                 
org.springframework.web.client.HttpClientErrorException$Unauthorized: 401 Unauthorized: "<html><EOL><EOL><head><title>401 Authorization Required</title></head><E
OL><EOL><body><EOL><EOL><center><h1>401 Authorization Required</h1></center><EOL><EOL><hr><center>Redirecting to <a id='link' href=''></a></center><EOL><EOL><scr
ipt><EOL><EOL>const redirect_url = window.location.protocol + '//' + window.location.host + '/home';<EOL><EOL>function redirect() { window.location.href = redire
ct_url; }<EOL><EOL>window.onload = function() {<EOL><EOL>    const link = document.getElementById('link');<EOL><EOL>    link.href = redirect_url;<EOL><EOL>    li
nk.text = redirect_url;<EOL><EOL>    setTimeout(redirect, 500);<EOL><EOL>}<EOL><EOL></script><EOL><EOL></body><EOL><EOL></html><EOL>"                            
        at org.springframework.web.client.HttpClientErrorException.create(HttpClientErrorException.java:105) ~[spring-web-5.3.23.jar:5.3.23]                     
        at org.springframework.web.client.DefaultResponseErrorHandler.handleError(DefaultResponseErrorHandler.java:168) ~[spring-web-5.3.23.jar:5.3.23]          
[... cut ...]
        at org.springframework.web.client.ResponseErrorHandler.handleError(ResponseErrorHandler.java:63) ~[spring-web-5.3.23.jar:5.3.23]                         
        at org.springframework.web.client.RestTemplate.handleResponse(RestTemplate.java:819) ~[spring-web-5.3.23.jar:5.3.23]                                     
        at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:777) ~[spring-web-5.3.23.jar:5.3.23]                                          
        at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:711) ~[spring-web-5.3.23.jar:5.3.23]

rice1204, 2023-07-10:
>  tried to generate a token through https://entrez.enphaseenergy.com/ but I get a page with Select System / Select Gateway after logging in. Both these fields are blanked and there's nothing in the drop down list. I am guessing that's something wrong with my Enphase account?

in the "select system" field, you have to type in the first 2-3 characters of your system name and then it will autocomplete (i have no idea why they hell they designed it this way). This will depend on installer, but in my case, my system name was my first name. After that, you'll get to select your gateway.

dlmcpaul, 2023-07-10:
Ok weird.  I got an email with a comment on this issue and @rice1204 managed to reply to it but it not here.

From the email I saw you look to be running version 0.26 and you need to be at least 0.27 as most reports have that working.

Probably a good time to summarise what we need to get the new authorisation working
Version 0.27
Then supply in the command
ENVOY_BEARER_TOKEN or
ENVOY_ENPHASE_WEB_USER and ENVOY_ENPHASE_WEB_PASSWORD

You may also need to set ENVOY_CONTROLLER_PORT=443

@rice1204 I think you got yours working can you post the command line you used?

dlmcpaul, 2023-07-10:
Side note: I know the jump from 0.26 to 0.27 can be a big one for those running the jar directly as the internal database needs to be converted or deleted it you don't care about it's history.

Happy to try and support you through conversion

lukegiuliani, 2023-08-03:
@ChadDa3mon did you resolve your issue? I also seem to be having the challenge with self signed certs, full logs:

```
2023-08-03 01:56:23.258  INFO 1 --- [           main] com.hz.services.LocalDBService           : Summarising Event table prior to 2023-08-03T00:00
2023-08-03 01:56:33.842  INFO 1 --- [nio-8080-exec-1] o.a.c.c.C.[Tomcat].[localhost].[/solar]  : Initializing Spring DispatcherServlet 'dispatcherServlet'
2023-08-03 01:56:33.842  INFO 1 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Initializing Servlet 'dispatcherServlet'
2023-08-03 01:56:33.847  INFO 1 --- [nio-8080-exec-1] o.s.web.servlet.DispatcherServlet        : Completed initialization in 4 ms
2023-08-03 01:57:22.346 ERROR 1 --- [   scheduling-1] com.hz.services.EnvoyService             : Failed to retrieve Solar stats. Exception was I/O error on GET request for "https://192.168.20.218/home.json": PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target; nested exception is javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target

org.springframework.web.client.ResourceAccessException: I/O error on GET request for "https://192.168.20.218/home.json": PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target; nested exception is javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:785) ~[spring-web-5.3.23.jar!/:5.3.23]
	at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:711) ~[spring-web-5.3.23.jar!/:5.3.23]
	at org.springframework.web.client.RestTemplate.getForEntity(RestTemplate.java:361) ~[spring-web-5.3.23.jar!/:5.3.23]
	at com.hz.services.EnvoyService.getSystemData(EnvoyService.java:67) ~[classes!/:271-RC]
	at com.hz.services.EnvoyService.collectEnphaseData(EnvoyService.java:78) ~[classes!/:271-RC]
	at com.hz.services.OutputManager.gather(OutputManager.java:45) ~[classes!/:271-RC]
	at jdk.internal.reflect.GeneratedMethodAccessor74.invoke(Unknown Source) ~[na:na]
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source) ~[na:na]
	at java.base/java.lang.reflect.Method.invoke(Unknown Source) ~[na:na]
	at org.springframework.scheduling.support.ScheduledMethodRunnable.run(ScheduledMethodRunnable.java:84) ~[spring-context-5.3.23.jar!/:5.3.23]
	at org.springframework.scheduling.support.DelegatingErrorHandlingRunnable.run(DelegatingErrorHandlingRunnable.java:54) ~[spring-context-5.3.23.jar!/:5.3.23]
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source) ~[na:na]
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source) ~[na:na]
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source) ~[na:na]
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source) ~[na:na]
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source) ~[na:na]
	at java.base/java.lang.Thread.run(Unknown Source) ~[na:na]
Caused by: javax.net.ssl.SSLHandshakeException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.ssl.Alert.createSSLException(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.TransportContext.fatal(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.TransportContext.fatal(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.TransportContext.fatal(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.checkServerCerts(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.onCertificate(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.CertificateMessage$T12CertificateConsumer.consume(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.SSLHandshake.consume(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.HandshakeContext.dispatch(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.HandshakeContext.dispatch(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.TransportContext.dispatch(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.SSLTransport.decode(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.SSLSocketImpl.decode(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.SSLSocketImpl.readHandshakeRecord(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(Unknown Source) ~[na:na]
	at java.base/sun.security.ssl.SSLSocketImpl.startHandshake(Unknown Source) ~[na:na]
	at org.apache.http.conn.ssl.SSLConnectionSocketFactory.createLayeredSocket(SSLConnectionSocketFactory.java:436) ~[httpclient-4.5.13.jar!/:4.5.13]
	at org.apache.http.conn.ssl.SSLConnectionSocketFactory.connectSocket(SSLConnectionSocketFactory.java:384) ~[httpclient-4.5.13.jar!/:4.5.13]
	at org.apache.http.impl.conn.DefaultHttpClientConnectionOperator.connect(DefaultHttpClientConnectionOperator.java:142) ~[httpclient-4.5.13.jar!/:4.5.13]
[... cut ...]
	at java.base/sun.security.ssl.X509TrustManagerImpl.checkServerTrusted(Unknown Source) ~[na:na]
	... 53 common frames omitted
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
	at java.base/sun.security.provider.certpath.SunCertPathBuilder.build(Unknown Source) ~[na:na]
	at java.base/sun.security.provider.certpath.SunCertPathBuilder.engineBuild(Unknown Source) ~[na:na]
	at java.base/java.security.cert.CertPathBuilder.build(Unknown Source) ~[na:na]
	... 58 common frames omitted
```

I'm running on docker-arm.

dlmcpaul, 2023-08-03:
@lukegiuliani sorry which docker image are you pulling?

The experimental tag for dlmcpaul/enphasecollector-arm is a little behind the release as it is about 8 months old.

I have pushed a more recent build to both x86 and arm experimental tags so you can test it out.  

Arm releases are a bit tricky these days as there are some older Pi machines that I can no longer support.

lukegiuliani, 2023-08-03:
Sorry I should have been more specific (and also thanked you for the awesome repo!)

`dlmcpaul/enphasecollector-arm:experimental` is what i'm currently using, after `dlmcpaul/enphasecollector-arm` had the same issue. 

But either way, the new build you just pushed has fixed it, thank you very much ❤️

dlmcpaul, 2023-08-03:
Ok good.  I really should cut a release.  I think I did something between the last release and the current experimental that handles some extra edge cases with the new firmware.

dlmcpaul, 2023-09-08:
I have done a new release with all the fixes identified including handling the recent change to the login API.

Will close this issue soon.  Please open a new issue for any bugs found.