dlmcpaul/EnphaseCollector #54: Update "How to build" docu

drache42, 2023-04-20:
It would be super helpful if you could update the `README.md` with the instructions on how to build/test/run this locally.

Some people, like me, have never used springboot or maven, so it would be very helpful if you could give quick step by step instructions on how I can build and run this locally. 

Thank you!

dlmcpaul, 2023-04-21:
I have updated the DOCKER.md file with additional information to build using docker.  If you don't have much maven experience then that may help but you will need to be able to install docker first and that can be more complex than installing java plus maven.

Are you looking to just run the code or interested in understanding or making changes for yourself?

drache42, 2023-04-21:
Thank you for your response.
I would like to make changes to the code, but I am not familiar with spring boot or maven. I'm a dotnet dev so the Java ecosystem is new to me.
I was able get it to compile in VSCode... I just don't understand how it did it yet :) 

I created a spring boot start thing, downloaded the .mvn folder, but when I type `.\mvnw.cmd compile` I get the following error, which I don't fully understand yet
```
[INFO] --- spring-boot-maven-plugin:3.0.5:build-info (default) @ enphasecollector ---
[WARNING] Error injecting: org.springframework.boot.maven.BuildInfoMojo
java.lang.TypeNotPresentException: Type org.springframework.boot.maven.BuildInfoMojo not present
    at org.eclipse.sisu.space.URLClassSpace.loadClass (URLClassSpace.java:147)
    at org.eclipse.sisu.space.NamedClass.load (NamedClass.java:46)
    at org.eclipse.sisu.space.AbstractDeferredClass.get (AbstractDeferredClass.java:48)
    at com.google.inject.internal.ProviderInternalFactory.provision (ProviderInternalFactory.java:85)
    at com.google.inject.internal.InternalFactoryToInitializableAdapter.provision (InternalFactoryToInitializableAdapter.java:57)
```

dlmcpaul, 2023-04-21:
Not sure you gave the complete error info but my first guess is that you are using an old java version.

Check what the value of JAVA_HOME is.  You need Java 17 or higher.

drache42, 2023-04-21:
I gave you the first error and part of the call stack. When I get home I'll
send you the entire thing.

I installed java 17 and made sure to set JAVA_HOME, else I wouldn't have
gotten this far :)

drache42, 2023-04-22:
Here is my entire output

```
❯ .\mvnw.cmd compile
[INFO] Scanning for projects...
[INFO]
[INFO] ----------------------< com.hz:enphasecollector >-----------------------
[INFO] Building EnphaseCollector development-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO]
[INFO] --- git-commit-id-maven-plugin:5.0.0:revision (default) @ enphasecollector ---
[INFO] dotGitDirectory D:\Code\EnphaseCollector\.git
[INFO] Collected git.commit.id with value befb0b34a3df2b8ce05696e3dc2ea310ac5741c5
[INFO] including property git.commit.id in results
[INFO] Reading existing properties file [D:\Code\EnphaseCollector\target\classes\git.properties] (for module EnphaseCollector)...
[INFO] Properties file [D:\Code\EnphaseCollector\target\classes\git.properties] is up-to-date (for module EnphaseCollector)...
[INFO]
[INFO] --- jacoco-maven-plugin:0.8.8:prepare-agent (default) @ enphasecollector ---
[INFO] argLine set to -javaagent:C:\\Users\\drach\\.m2\\repository\\org\\jacoco\\org.jacoco.agent\\0.8.8\\org.jacoco.agent-0.8.8-runtime.jar=destfile=D:\\Code\\EnphaseCollector\\target\\jacoco.exec
[INFO]
[INFO] --- spring-boot-maven-plugin:3.0.5:build-info (default) @ enphasecollector ---
[WARNING] Error injecting: org.springframework.boot.maven.BuildInfoMojo
java.lang.TypeNotPresentException: Type org.springframework.boot.maven.BuildInfoMojo not present
    at org.eclipse.sisu.space.URLClassSpace.loadClass (URLClassSpace.java:147)
    at org.eclipse.sisu.space.NamedClass.load (NamedClass.java:46)
    at org.eclipse.sisu.space.AbstractDeferredClass.get (AbstractDeferredClass.java:48)
    at com.google.inject.internal.ProviderInternalFactory.provision (ProviderInternalFactory.java:85)
    at com.google.inject.internal.InternalFactoryToInitializableAdapter.provision (InternalFactoryToInitializableAdapter.java:57)
    at com.google.inject.internal.ProviderInternalFactory$1.call (ProviderInternalFactory.java:66)
    at com.google.inject.internal.ProvisionListenerStackCallback$Provision.provision (ProvisionListenerStackCallback.java:112)
    at com.google.inject.internal.ProvisionListenerStackCallback$Provision.provision (ProvisionListenerStackCallback.java:127)
    at com.google.inject.internal.ProvisionListenerStackCallback.provision (ProvisionListenerStackCallback.java:66)
    at com.google.inject.internal.ProviderInternalFactory.circularGet (ProviderInternalFactory.java:61)
    at com.google.inject.internal.InternalFactoryToInitializableAdapter.get (InternalFactoryToInitializableAdapter.java:47)
    at com.google.inject.internal.InjectorImpl$1.get (InjectorImpl.java:1050)
    at org.eclipse.sisu.inject.Guice4$1.get (Guice4.java:162)
    at org.eclipse.sisu.inject.LazyBeanEntry.getValue (LazyBeanEntry.java:81)
    at org.eclipse.sisu.plexus.LazyPlexusBean.getValue (LazyPlexusBean.java:51)
    at org.codehaus.plexus.DefaultPlexusContainer.lookup (DefaultPlexusContainer.java:263)
    at org.codehaus.plexus.DefaultPlexusContainer.lookup (DefaultPlexusContainer.java:255)
    at org.apache.maven.plugin.internal.DefaultMavenPluginManager.getConfiguredMojo (DefaultMavenPluginManager.java:520)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:124)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute2 (MojoExecutor.java:370)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute (MojoExecutor.java:351)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:215)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:171)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:163)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:298)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:960)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:293)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:196)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:566)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:282)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
[... cut ...]
[ERROR] import: Entry[import  from realm ClassRealm[project>com.hz:enphasecollector:development-SNAPSHOT, parent: ClassRealm[maven.api, parent: null]]]
[ERROR]
[ERROR] -----------------------------------------------------
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginContainerException
```

dlmcpaul, 2023-04-22:
**java.lang.UnsupportedClassVersionError: org/springframework/boot/maven/BuildInfoMojo has been compiled by a more recent version of the Java Runtime (class file version 61.0), this version of the Java Runtime only recognizes class file versions up to 55.0**

So the above line says that the java compiler being used is Java 11 (Class files version 55) and you need Java 17 (Class files version 61).

Because I am using spring boot 3 all the spring libraries have been built with Java 17.

If JAVA_HOME is set to a Java 17 JDK maybe the path is set to a Java 11 version?

drache42, 2023-04-22:
Error between chair and keyboard it turns out. I had installed the wrong version of Java. 

It all works now. Thank you very much!

closed 2023-04-22 by drache42