dlmcpaul/EnphaseCollector #58: Dockerfile bug: unknown instruction: IF

drache42, 2023-04-23:
```
❯ docker build -f Dockerfile-build .
[+] Building 0.1s (2/2) FINISHED
 => [internal] load build definition from Dockerfile-build                                                                                                                                                                                                                         0.0s
 => => transferring dockerfile: 1.98kB                                                                                                                                                                                                                                             0.0s
 => [internal] load .dockerignore                                                                                                                                                                                                                                                  0.0s
 => => transferring context: 2B                                                                                                                                                                                                                                                    0.0s
failed to solve with frontend dockerfile.v0: failed to create LLB definition: dockerfile parse error line 27: unknown instruction: IF
```

dlmcpaul, 2023-04-23:
Hmm, try adding to the top of the docker file the line below.  It should force docker to use the newer syntax
# syntax=docker/dockerfile:1.4

drache42, 2023-04-23:
That fixes it!

dlmcpaul, 2023-04-24:
Ok, committed the changes

closed 2023-04-24 by dlmcpaul