dlmcpaul/EnphaseCollector #59: How to connect to MQTT publisher

rice1204, 2023-05-05:
I'd love to get this connected to my MQTT setup but I'm having a hard time locating the syntax for passing mqtt configuration parameters (mqtt host, username, pass, topic, etc).

Can anyone please point me in the right direction?

dlmcpaul, 2023-05-05:
The test case I use is here https://github.com/dlmcpaul/EnphaseCollector/blob/master/src/test/java/com/hz/MqqtTest.java

So the basic config params are:

envoy.mqqtResource.host
envoy.mqqtResource.port
envoy.mqqtResource.topic

It is a standard resource object so there is also:

envoy.mqqtResource.user
envoy.mqqtResource.password
envoy.mqqtResource.publisherId

I have not tested much beyond connecting to a MQQT server

The publisherId will default to a random uuid on each run if not set.

The metrics I am publishing are here https://github.com/dlmcpaul/EnphaseCollector/blob/master/src/main/java/com/hz/services/MqttService.java but I don't know if they fit any MQQT standard for metrics.

dlmcpaul, 2023-09-08:
@rice1204 If my response above has answered your question please close this issue.

rice1204, 2023-09-08:
Closed, thanks!

closed 2023-09-08 by rice1204