dlmcpaul/EnphaseCollector #62: Error retrieving solar stats - IO error

rkokfhung, 2023-07-10:
Hi all,

Been a user of the project for years so very appreciative of the team here!

My envoy got upgraded and I finally got the authentication via token working. I am now getting the following I/O error on GET request. Help anyone?

Thanks in advance.
Richard

2023-07-10T07:03:17.808Z  INFO 8 --- [   scheduling-1] com.hz.services.EnvoyConnectionProxy     : Preparing Realm Authentication Provider with user envoy        
2023-07-10T07:03:17.813Z  INFO 8 --- [           main] com.hz.EnphaseCollectorApplication       : Started EnphaseCollectorApplication in 19.629 seconds (process 
running for 20.706)                                                                                                                                              
2023-07-10T07:03:17.826Z  INFO 8 --- [   scheduling-1] com.hz.services.EnvoyConnectionProxy     : Reading from protected Envoy controller endpoint http://envoy.l
ocal/api/v1/production/inverters                                                                                                                                 
2023-07-10T07:03:18.427Z  INFO 8 --- [   scheduling-1] o.a.h.c.h.i.c.HttpRequestRetryExec       : Recoverable I/O exception (java.net.UnknownHostException) caugh
t when processing request to {}->http://envoy.local:80                                                                                                           
2023-07-10T07:03:18.429Z  INFO 8 --- [   scheduling-1] o.a.h.c.h.i.c.HttpRequestRetryExec       : Recoverable I/O exception (java.net.UnknownHostException) caugh
t when processing request to {}->http://envoy.local:80                                                                                                           
**_**2023-07-10T07:03:18.432Z ERROR 8 --- [   scheduling-1] com.hz.services.EnvoyService             : Failed to retrieve Solar stats. Exception was I/O error on GET 
request for "http://envoy.local/home.json": envoy.local                                                                                                          
                                                                                                                                                                 
org.springframework.web.client.ResourceAccessException: I/O error on GET request for "http://envoy.local/home.json": envoy.local**_**                                 
        at org.springframework.web.client.RestTemplate.createResourceAccessException(RestTemplate.java:888) ~[spring-web-6.0.8.jar:6.0.8]                        
        at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:868) ~[spring-web-6.0.8.jar:6.0.8]                                            
        at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:764) ~[spring-web-6.0.8.jar:6.0.8]                                              
        at org.springframework.web.client.RestTemplate.getForEntity(RestTemplate.java:405) ~[spring-web-6.0.8.jar:6.0.8]                                         
        at com.hz.services.EnvoyService.getSystemData(EnvoyService.java:46) ~[app/:na]                                                                           
        at com.hz.services.EnvoyService.collectEnphaseData(EnvoyService.java:57) ~[app/:na]                                                                      
        at com.hz.services.OutputManager.gather(OutputManager.java:47) ~[app/:na]                                                                                
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[na:na]                                                               
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:77) ~[na:na]                                             
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[na:na]                                     
        at java.base/java.lang.reflect.Method.invoke(Method.java:568) ~[na:na]                                                                                   
        at org.springframework.scheduling.support.ScheduledMethodRunnable.run(ScheduledMethodRunnable.java:84) ~[spring-context-6.0.8.jar:6.0.8]                 
        at org.springframework.scheduling.support.DelegatingErrorHandlingRunnable.run(DelegatingErrorHandlingRunnable.java:54) ~[spring-context-6.0.8.jar:6.0.8] 
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539) ~[na:na]                                                            
        at java.base/java.util.concurrent.FutureTask.runAndReset(FutureTask.java:305) ~[na:na]                                                                   
        at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:305) ~[na:na]                     
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136) ~[na:na]                                                    
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635) ~[na:na]                                                    
        at java.base/java.lang.Thread.run(Thread.java:833) ~[na:na]                                                                                              
Caused by: java.net.UnknownHostException: envoy.local                                                                                                            
        at java.base/java.net.InetAddress$CachedAddresses.get(InetAddress.java:801) ~[na:na]                                                                     
        at java.base/java.net.InetAddress.getAllByName0(InetAddress.java:1533) ~[na:na]                                                                          
        at java.base/java.net.InetAddress.getAllByName(InetAddress.java:1385) ~[na:na]

dlmcpaul, 2023-07-10:
Are you using the jar or docker image?

rkokfhung, 2023-07-10:
> Are you using the jar or docker image?

Hi @dlmcpaul  I am using the docker image, running on my Synology NAS. I've used your project for years until this new API auth upgrade. By the way many thanks for your great work on this repo!

dlmcpaul, 2023-07-10:
Ok I have pushed an experimental image dlmcpaul/enphasecollector:experimental you can try

For V7 you will need to supply token or web user details from your enphase website login.
ENVOY_BEARER_TOKEN or
ENVOY_ENPHASE_WEB_USER and ENVOY_ENPHASE_WEB_PASSWORD

rkokfhung, 2023-07-10:
> Ok I have pushed an experimental image dlmcpaul/enphasecollector:experimental you can try
> 
> For V7 you will need to supply token or web user details from your enphase website login. ENVOY_BEARER_TOKEN or ENVOY_ENPHASE_WEB_USER and ENVOY_ENPHASE_WEB_PASSWORD

Thanks @dlmcpaul, I just stepped out of the house at the moment Will try the new image as soon as I get home. Thanks for replying promptly!

rkokfhung, 2023-07-10:
> Ok I have pushed an experimental image dlmcpaul/enphasecollector:experimental you can try
> 
> For V7 you will need to supply token or web user details from your enphase website login. ENVOY_BEARER_TOKEN or ENVOY_ENPHASE_WEB_USER and ENVOY_ENPHASE_WEB_PASSWORD

Awesome @dlmcpaul ! It worked! I just downloaded the experimental release and re-entered all the env variables and voila! Thanks very much for the quick turn around and helping me resolve the issue! Please let me know when this is merged to the latest release.

(By the way I used ENVOY_BEARERTOKEN per README instead of ENVOY_BEARER_TOKEN. It worked so I didn't try the latter)

Thanks again!

rkokfhung, 2023-07-10:
Issue resolved with experimental release for docker

dlmcpaul, 2023-07-10:
Ok, I do want to do a release.  Mostly minor bug fixes but I want to update to the latest Spring Boot 3.1 line as well so we will see when I can do it.